JoelBender/BACpypes3 #11: not responding to who-is

bbartling, 2023-09-27:
Hi Joel,

Am testing out `[mini_device_revisited](https://gist.github.com/bbartling/573030a2facc97e178b2362606b9cdda#file-mini_device_revisited-py).py` just on an Ubuntu x86 its a virtual machine that I run a lot of BAC0 apps on.

In terminal: 
```
$ python3.9 mini_device_revisited.py --debug bacpypes3.ipv4.IPv4DatagramProtocol
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:connection_made <_SelectorDatagramTransport fd=6 read=idle write=<idle, bufsize=0>>
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:    - socket: <asyncio.TransportSocket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=17, laddr=('10.200.200.223', 47808)>
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:    - socket_name: ('10.200.200.223', 47808)
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81\n\x00%\x01\x04\x02D0\x0e\x0c\x00\x80\x00\x01\x1e\tU\x1f\x0c\x01@\x00\x00\x1e\tU\x1f\x0c\x00\x80\x00\x00\x1e\tU\x1f' ('10.200.200.166', 47808)
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81\n\x00%\x01\x04\x02D\xd5\x0e\x0c\x00\x80\x00\x01\x1e\tU\x1f\x0c\x01@\x00\x00\x1e\tU\x1f\x0c\x00\x80\x00\x00\x1e\tU\x1f' ('10.200.200.166', 47808)
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81\n\x00%\x01\x04\x02Dy\x0e\x0c\x00\x80\x00\x01\x1e\tU\x1f\x0c\x01@\x00\x00\x1e\tU\x1f\x0c\x00\x80\x00\x00\x1e\tU\x1f' ('10.200.200.166', 47808)
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81\n\x00%\x01\x04\x02D\x1f\x0e\x0c\x00\x80\x00\x01\x1e\tU\x1f\x0c\x01@\x00\x00\x1e\tU\x1f\x0c\x00\x80\x00\x00\x1e\tU\x1f' ('10.200.200.166', 47808)
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81\n\x00%\x01\x04\x02D\xc3\x0e\x0c\x00\x80\x00\x01\x1e\tU\x1f\x0c\x01@\x00\x00\x1e\tU\x1f\x0c\x00\x80\x00\x00\x1e\tU\x1f' ('10.200.200.166', 47808)
```

Am running Contemporary Controls BACnet discover tool from a Windows OS to do the `who-is` but the bacpypes3 app isnt coming thru... any ideas to try?

`10.200.200.166` is VOLTTRON edge device trying to poll a BAC0 app that isnt running.

Funny thing is that this works just fine on a Rasp pi in docker with these steps:
https://gist.github.com/bbartling/573030a2facc97e178b2362606b9cdda

JoelBender, 2023-09-28:
This is because your application is not listening on the broadcast address, and I can tell that because it has only opened one socket for unicast messages.  Try adding `--address 10.200.200.223/24` and notice that it will be listening on 10.200.200.255, and you'll need to adjust that for the size of your subnet so check the number of 1's in your network mask.

If you have the [ifaddr](https://pypi.org/project/ifaddr/) library installed, BACpypes3 will use that to determine the IPv4 address and subnet so you won't need to provide it which is handy in DHCP environments.  If you have multi-homing (multiple IPv4 addresses on different interfaces, or multiple VLANs on the same interface) you'll need to give it some help by providing the interface name like `--address enp0s25`.

JoelBender, 2023-09-28:
Be aware that some virtual machine environments (VMware Fusion, VirtualBox, KVM, Parallels, etc.) are similar to docker in that they set up a virtual network inside the host that could be disconnected from the real world.  It's nice for testing client/server communications without a lot of "noise."

JoelBender, 2023-09-28:
Oh, one more thing; if you're running an image in a docker container on Windows, it will not allow the container to open a socket on the broadcast address and you'll have to register with some BBMD on your network as a foreign device.  There is probably a setting someplace to unblock this.

bbartling, 2023-09-28:
Thanks Joel!

bbartling, 2023-09-29:
Hey Joel,

Am getting a bit deeper into the weeds to troubleshoot why the [mini_device_revisted.py](https://gist.github.com/bbartling/573030a2facc97e178b2362606b9cdda#file-mini_device_revisited-py) isn't responding to a `who-is` on my home test bench LAN when I run it from terminal but it does on Docker with a Dockerfile stating:

`# Expose the UDP port 47808
EXPOSE 47808/udp`

and running it with: `sudo docker run -d -p 47808:47808/udp --name my-bacpypes-container bacpypes-three-mini-device-revisited`

Okay so trying this...no Docker just running the `mini_device_revisted.py` file from terminal on rasp pi python 3.9 - with this added method and nothing no response or any `debug` prints for  `_log.debug("Received WhoIsRequest")`.

```
from bacpypes3.service.device import WhoIsRequest, IAmRequest

    async def do_WhoIsRequest(self, request):
        if _debug:
            _log.debug("Received WhoIsRequest")
        response = IAmRequest(
            iAmDeviceIdentifier=self.thisDevice.objectIdentifier,
            maxAPDULengthAccepted=self.maxApduLengthAccepted,
            segmentationSupported=self.segmentationSupported,
            vendorID=self.vendorIdentifier,
        )
        self.response(response)
```

From a different SSH terminal the rasp pi appears to be listening when mini_device_revisted.py is running:
```
bbartling@benspi:~ $ sudo netstat -tuln | grep 47808
udp        0      0 192.168.0.101:47808     0.0.0.0:*
```

Am getting out of my comfort zone fast here so any tips appreciated...trying to make pcap with tcpdump for the `who-is` sent from my Windows computer the rasp pi did see 4 packets that I can make out:
```
$ sudo tcpdump -i eth0 -w bens_pcap.pcap udp port 47808
tcpdump: listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
^C4 packets captured
4 packets received by filter
0 packets dropped by kernel
```

after getting the pcap file back to my Windows computer which i did the `who-is` from on the Contemporary Controls BACnet discovery tool...which a few old MSTP device did respond those are the  devices 201201 and 202202 and the BACnet router is device 0...and my Windows machine is 192.168.0.109.

[image]

So in my very limited experience of doing pcaps, the rasp pi I think did see the `who-is`.

Any ideas to try? Am learning a ton here! Both in asyncio and pcaps! Thanks for anytime to respond!

bbartling, 2023-10-01:
`pip install ifaddr` and it should work!!

At least its working for me on Rasp pi B+ Model running latest Raspbian OS with Python 3.9

closed 2023-10-01 by bbartling