JoelBender/BACpypes3 #18: read_property_multiple API update

JoelBender, 2023-10-16:
The `read_property_multiple()` parameter list is weird and not intuitive.  Try something else.

JoelBender, 2023-10-16:
`PropertyReference` can now be constructed with a string like `"object-name"`, a string with an array index like `"object-list[1]"`, a string of digits like `"7775"`, a string with an array index like `"7775[2]"`, an integer like `7775`, or a tuple like `("object-list", 2)`, or `(7775, 2)`.  Lots of options.

It also takes an optional `vendor_identifier` keyword argument so that a string like `"snork"` can reference proprietary property names.

JoelBender, 2023-10-16:
The `PropertyIdentifier` class definition had to move from **basetypes** to **primitivedata** so it could be imported by **vendor**.

ChristianTremblay, 2023-10-17:
some work done here : https://github.com/ChristianTremblay/BACpypes3/tree/issue18

JoelBender, 2023-10-18:
@ChristianTremblay try this branch with your fork

ChristianTremblay, 2023-10-18:
@JoelBender Regading `hash(id(self))`. I tried to remove hash and I got this : 

```
Checking 00020813-0000-0000-C000-000000000046x0x1x9
Could not add module (IID('{00020813-0000-0000-C000-000000000046}'), 0, 1, 9) - <class 'ImportError'>: cannot import name 'Dispatch' from partially initialized module 'win32com.client' (most likely due to a circular import) (C:\Users\ctremblay\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\win32com\client\__init__.py
```

So I kept it in

closed 2023-10-20 by JoelBender