JoelBender/BACpypes3 #25: unable to get the ALARM during the runtime of application

BMPC2023, 2024-02-05:
Hi @JoelBender Sir,
Thanks for BACpypes3, for the new feature support I have migrated my code from bacpypes to bacpypes3 where all the basic read,write and COV works fine but I am facing issue for not getting the ALARM (Yabe Status : ALARM) , while updating the presentValue in the code level when the app is running. 

and using Yabe if I try to write the presentValue explicitly then I am getting the ALARM for above/below limit mentioned.

I have taken an example code from [https://github.com/JoelBender/BACpypes3/blob/main/sandbox/event-server-avo1-ir.py](here)
 
So please can you suggest me this...

JoelBender, 2024-02-06:
When you say "in the code level" are you referring to [this statement](https://github.com/JoelBender/BACpypes3/blob/main/sandbox/event-server-avo1-ir.py#L136) in the sample?  There is a difference between setting an attribute like `thing.presentValue = 12.3` which is `setattr(thing, "presentValue", 12.3)` and writing a value like `thing.write_property("presentValue", 12.3)` because sometimes you want there to be a difference, but it's not very clear.

This intrinsic reporting code doesn't have any tests, and there's something about transitioning to/from fault that with notification delays that doesn't sit well with me yet, so that's why it's sitting the sandbox.

BMPC2023, 2024-02-06:
The Above three formats for setting the presentValue I have tried sir, values are updating properly but the status of flag is not changing with respect to crossing the above /below limit sir (during runtime). 

I have taken the sample for Sandbox for the external feature(ALARM) support to my project sir, So if their is any other samples to refer,please let me know sir

JoelBender, 2024-02-17:
For part (1) I'll figure out why it's not working.

For part (2) there is a mismatch in the API for the `DeviceAddress` class which has been fixed in [this commit](https://github.com/JoelBender/BACpypes3/compare/main...26-custom-router-info) which is going to get rolled into the next release.  The acceptable forms are:

```python
>>> Recipient(address=b'123456').debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7aac1ebc0>
        macAddress = b'123456'

>>> Recipient(address="1.2.3.4").debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7abc9fcd0>
        networkNumber = 0
        macAddress = b'\x01\x02\x03\x04\xba\xc0'

>>> Recipient(address=Address("1.2.3.4")).debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7abc9e260>
        networkNumber = 0
        macAddress = b'\x01\x02\x03\x04\xba\xc0'

>>> Recipient(address=Address("5:6.7.8.9")).debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7abbe5d50>
        networkNumber = 5
        macAddress = b'\x06\x07\x08\t\xba\xc0'

>>> Recipient(address=DeviceAddress("10:11")).debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7abcadb40>
        networkNumber = 10
        macAddress = b'\x0b'

>>> Recipient(address=DeviceAddress(networkNumber=12, macAddress=b'\x0d')).debug_contents()
    address = <bacpypes3.primitivedata.DeviceAddress object at 0x7fc7aac1ec20>
        networkNumber = 12
        macAddress = b'\r'
```

JoelBender, 2024-02-17:
I updated the [sample application](https://github.com/JoelBender/BACpypes3/blob/8672dbc4134bfa5af6228cb3094f606a0969e973/sandbox/event-server-avo1-ir.py) to add a `recipient` command line argument and `--confirmed` option.  The recipient is the BACnet address where the notifications should go and confirmed is to issue confirmed or unconfirmed notifications.

I think the reason that you're not getting the notifications is stumbling around the dynamic binding in the sample (where the `Recipient` was given the object identifier) and the static binding the updated sample uses.  Between these two updates it should work.

BMPC2023, 2024-02-18:
Hi @JoelBender Sir,
Thanks for the reply and suggestionsSir, So that I able to implement the IR (Internsic reporting for the AnalogInput and AnalogOutput Object).

 And I have some doubts regarding the ALARM services for BinaryInput and BinaryOutput Object,where i have tried the Internsic reporting using the this [https://github.com/JoelBender/BACpypes3/blob/8672dbc4134bfa5af6228cb3094f606a0969e973/sandbox/event-server-bvo1.py](url) but I the status is not changes for the mentioned AlarmValue (BI object) and feedbackValue (BO obj),So please let me know sir the correct format to implement the Internsic reporting for BinaryInput and BinaryOutput Object.

JoelBender, 2024-02-18:
Notice that there is no code in the `execute()` method of the `CommandFailureEventAlgorithm`, that has not been implemented so Binary Output and Multi-state Output objects aren't going to get intrinsic reporting until it is.  I updated the binary value object samples with the same command line recipient argument as the analog value sample.

shashankgowdasd, 2024-02-18:
Sir for Binaryinput, algorithm used is ChangeOfStateEventAlgorithmand and for BinaryOutput CommandFailueEventAlgorithm .So for BinaryInput obj Internsic reporting that should work for the same format with change in alarm value property sir but the state and notification is not working in my case sir,

So please let me know where I have to correct it to get the change of state and notification to recipient for BinaryInput object sir.

JoelBender, 2024-02-19:
I cloned the binary value object sample and replaced "Value" with "Input" and it seems to work the same, sandbox code is [here](https://github.com/JoelBender/BACpypes3/blob/26-custom-router-info/sandbox/event-server-bio1.py).

BMPC2023, 2024-02-20:
Hi @JoelBender Sir,
I tried using the [https://github.com/JoelBender/BACpypes3/blob/26-custom-router-info/sandbox/event-server-bio1.py](url) for binaryInput with cmd as  

> python event-server-bvo1.py 192.168.0.119 --confirmed

After that I updated the presentValue of binaryInput object using "Yabe", for to get the notification but I didn't Sir.

But for the binaryValue object , I am able to get the notification perfectly sir, So let me know the correct way for to implement the binaryInput object internsic reporting Sir.

BMPC2023, 2024-02-23:
Hi @JoelBender Sir,
Thanks for the Sandbox code of BinaryInput object and it works with me.

But at the time of routing notification to the recipient through the objects like AnalogInput, AnalogOutput and BinaryInput objects I am getting an below Runtime Error Sir, So I want to know where I am going wrong sir

> Task exception was never retrieved
future: <Task finished name='Task-901' coro=<IPv4DatagramServer.confirmation() done, defined at C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\ipv4\__init__.py:232> exception=RuntimeError('Event loop is closed')>
Traceback (most recent call last):
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\ipv4\__init__.py", line 243, in confirmation       
    await self.response(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\ipv4\service.py", line 115, in confirmation        
    await self.annexJ.response(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\ipv4\bvll.py", line 252, in confirmation
    await self.response(lpdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\ipv4\service.py", line 333, in confirmation        
    await self.response(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\netservice.py", line 367, in confirmation
    await self.adapterSAP.process_npdu(self, npdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\netservice.py", line 886, in process_npdu
    await self.response(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\appservice.py", line 1669, in confirmation
    await tr.confirmation(apdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\appservice.py", line 593, in confirmation
    await self.await_confirmation(apdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\appservice.py", line 772, in await_confirmation    
    await self.response(apdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\appservice.py", line 582, in response
    await self.ssmSAP.sap_response(apdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\appservice.py", line 1838, in sap_response
    await ServiceAccessPoint.sap_response(self, xpdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\comm.py", line 187, in sap_response
    await self.serviceElement.confirmation(*args)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\app.py", line 984, in confirmation
    future.set_result(apdu)
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 746, in call_soon
    self._check_closed()
  File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 510, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed