JoelBender/BACpypes3 #27: Format to set the Recipient Address of NotificationClass Object.

BMPC2023, 2024-02-16:
Hi @JoelBender Sir,
Is their any proper format for setting the IPv4 address in the Recipient Address of notificationClass Object.

And I have tried the following structure sir but I am getting the Error 

 

> 
       NotificationClassObject(
            objectIdentifier="notification-class,1",
            objectName="nc1",
            description="test notification class",
            notificationClass=1,
            priority=[9, 9, 9],  # toOffNormal, toFault, toNormal priority
            ackRequired=[0, 0, 0],  # toOffNormal, toFault, toNormal
            recipientList=[
                Destination(
                    validDays=[1, 1, 1, 1, 1, 1, 1],
                    fromTime=(0, 0, 0, 0),
                    toTime=(23, 59, 59, 99),
                    # recipient=Recipient(address=DeviceAddress(networkNumber=0,macAddress=Address("192.0.168.189").addrAddr)),
                    # recipient=Recipient(device="device,990",address=DeviceAddress(0,b"\x48\xE7\xDA\x95\x94\x63")),            
                    recipient=Recipient(device="device,990",address=DeviceAddress(networkNumber=255,
                                                         macAddress=IPv4Address("192.0.168.189").addrAddr)),
                    processIdentifier=0,
                    issueConfirmedNotifications=True,
                    transitions=[1, 1, 1],  # toOffNormal, toFault, toNormal
                )
            ],
        )

I have tried 3 formats in which all gave error 

Error:

  

> File "c:\Users\vimel\SHASHANK_S_D\PDF\BACnet\BACpypes3_updated_27_12_2023\BACpypes3-main\sandbox\event-server-avo1-ir.py", line 288, in main
>     recipient=Recipient(device="device,990",address=DeviceAddress(networkNumber=255,macAddress=IPv4Address("192.0.168.189").addrAddr)),      
>   File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\constructeddata.py", line 228, in __call__
>     return cast(Sequence, type.__call__(new_type, *args, **kwargs))
>   File "C:\Users\vimel\AppData\Local\Programs\Python\Python39\lib\site-packages\bacpypes3\constructeddata.py", line 630, in __init__
>     raise RuntimeError("initialize one choice: " + ", ".join(kw_list))
> RuntimeError: initialize one choice: device, address

JoelBender, 2024-02-16:
The `Recipient` is a choice between the `device` referencing an object identifier of a device object (in which case the application will need to resolve it) _or_ an `address`, commonly referred to as a form of "static binding".  You have provided both.

But beside that, there is a different problem with converting a `DeviceAddress` that is a version that is context encoded.  The fix for it is stacked up behind a separate problem, so in the mean time please use the form `Recipient(address="255:192.0.168.189")` and that will set both the `networkNumber` and `macAddress` correctly.

JoelBender, 2024-02-17:
See update and original in issue #25

closed 2024-02-17 by JoelBender