JoelBender/BACpypes3 #30: Creating an Address by interface name fails if the interface is a bridge interface

mikran, 2024-02-26:
We had a bridge interface with a name `br-ifLan1` which leads to `ValueError("invalid address")`.

This regex is restrictive on what interface names can be passed to Addresses: https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/pdu.py#L101

JoelBender, 2024-02-27:
What do you think about this pattern?  It should at least start with a letter.
```python
interface_port_re = re.compile(r"^(?:([A-Za-z][\w.-]*))(?::(\d+))?$")
```

mikran, 2024-02-27:
How about using `netifaces` if available and then having that regex as a fallback?

JoelBender, 2024-02-28:
`netifaces` doesn't have a maintainer so it should be considered deprecated and I've switched to `ifaddr`.  The API's are different enough that it makes the other code that follows a bit tedious.  If there was just one to support it wouldn't be too bad to refactor the rest of the code, but to do that is stacked up behind some other work.  I'll adjust the expression for now to get over the immediate problem.

mikran, 2024-02-28:
Oh good to know, I saw that the same file had soft dependency to both so I installed both but then I'll want to drop netifaces. I guess this issue is solved already.

closed 2024-02-28 by mikran