JoelBender/BACpypes3 #33: Software license

obscuresc, 2024-03-05:
Hi,

I noticed that there is no software license listed on BACpypes3. 
Could one be added so I know what we can do with the project?

Thanks!

JoelBender, 2024-03-30:
@obscuresc Is there one you would recommend?  I'm considering the example [here](https://github.com/bacnet-stack/bacnet-stack/blob/master/license/readme.txt) used by @skarg.

zoopp, 2024-03-31:
I apologize for intervening, but ultimately, the decision should largely rests on how you, as the author, want to approach it. You might find the following resources helpful:

- [The Open Source Guide on Legal Matters](https://opensource.guide/legal)
- [Choose a License](https://choosealicense.com/)
    - [Chose a License: Licenses](https://choosealicense.com/licenses/)
    - [Chose a License: Appendix](https://choosealicense.com/appendix/)

The first link is about why having a license matters and the next links give a good TL/DR overview + additional details on what some of the most common open source licenses are about.

JoelBender, 2024-03-31:
@zoopp Thank you!  Off to reading...

lukeyeager, 2024-08-05:
+1 FWIW, once you pick one I can [probably] start using this and I'll file bug reports for any issues I run into.

Sent here by https://github.com/JoelBender/bacpypes/issues/525#issuecomment-2106248400 (thanks!).

lukeyeager, 2024-08-05:
Did you intend to pick BSD in setup.py? https://github.com/JoelBender/BACpypes3/blob/v0.0.97/setup.py#L33