JoelBender/BACpypes3 #42: Value error minimum length: 1

xkostrna, 2024-05-14:
Hi, I'm getting error on line **1392** in **primitivedata.py** file inside **CharacterString** class, even after further investigation I wasn't able to figure out why is this happening. I would be really happy if you could point me out to finding the root cause.

More info:
It keeps happening when I'm trying to use **Application.read_property_multiple()** method with **parameter_list** of 200 elements.

JoelBender, 2024-05-15:
Object names have a minimum character length of one character, I suspect that the response that you are getting back from the device you are reading from isn't configured correctly.

xkostrna, 2024-05-15:
Yes, your suggestion is on point. We resolved the issue yesterday and actually found that we were trying to read property called **Object_Name** which was actually empty and since your code follows standard which says that this property must have at least one character we got an error. However I think it would be more appropriate to somehow specify which property was problematic in exception message. I wasn't able to figure out until I put some prints into your code.

JoelBender, 2024-05-15:
Good idea, labeling this issue as an enhancement request and leaving it open.