JoelBender/BACpypes3 #43: BACnet Routing

kheldaroz402, 2024-05-15:
HI Joel & Crew,

I have a building controller with two network ports, eth0 and wlan0.
What I am hoping is possible, is to be able to use the wifi (wlan0) as an access point - setup and working can access all other controllers, internet etc. to be able to use yabe to view/modify points on the eth0 network.
[image]

Since I have a bacpypes application (not 3) running on eth0 already, I bound a second IP address to eth0 so that I could run a router on the other IP address and thus be able to see the BACnet data on my laptop.

Is this even possible ?
Is this the best method ?

Which script would be best to use as a base ?
I've tried -  bacpypes3 

- ipv4-to-ipv4.py
- router-json.py

I've also tried - bacpypes

- router
- nat
scripts

wireshark - running on both my PC (connected to 192.168.2.xx range) and laptop (connected to 192.168.7.xx) range

yabe on PC finds my BACnet Device
yabe on laptop finds nothing

should a bacnet router show up in yabe ?

JoelBender, 2024-05-16:
Single link dual-homing like this so you can run two BACnet applications/stacks on the same machine is an interesting solution!  Note that you can only have **one** BACnet router between `192.168.3.0/24` and `192.168.7.0/24`, the other two PCs are just regular BACnet devices (all three PCs can't all have the same wlan0 address).

The `router-json.py` sample application should work just fine as an IPv4-to-IPv4 router, there are just a few things in the `router-config.json` file that need site-specific values.  You probably didn't notice that the sample configuration file is for one IPv4 address but two different ports.
```
{
    ...
    "application": [
        {
            ...
            "object-identifier": "device,999",  # unique instance number on the BACnet intranet
            "object-name": "My Router",  # also unique
            ...
        },
        {
            ...
            "ip-address": "192.168.3.41",  # address of PC on eth0
            "ip-subnet-mask": "255.255.255.0",
            ...
            "mac-address": "192.168.3.41:47808",
            "network-number": 100,
            ...
            "object-identifier": "network-port,1",  # unique within the device
            "object-name": "NetworkPort-1",  # also unique
            ...
        },
        {
            ...
            "ip-address": "192.168.7.1",  # address of PC on wlan0
            "ip-subnet-mask": "255.255.255.0",
            ...
            "mac-address": "192.168.7.1:47808",  # same as above, standard port
            "network-number": 200,
            ...
            "object-identifier": "network-port,2",
            "object-name": "NetworkPort-2",
            ...
        }
    ]
}
```

kheldaroz402, 2024-05-16:
The other devices (RPI) won't be on the same "network" even though they have the same ip range (192.168.7.1), the idea is that they will be in different plant rooms and will allow technicians to log into the devices via a WiFi AP without having to plug a cable into the controller

I did miss changing one of the ports but still no success

this is from another device on the 192.168.2.XX range

`{
    "BACpypes": {
        "backup_count": 5,
        "color": false,
        "config": {},
        "cov_lifetime": 60,
        "debug": [],
        "debug_file": "",
        "max_bytes": 1048576,
        "route_aware": false
    },
    "application": [
        {
            "active-cov-subscriptions": [],
            "apdu-segment-timeout": 1000,
            "apdu-timeout": 3000,
            "application-software-version": "1.0",
            "database-revision": 1,
            "device-address-binding": [],
            "firmware-revision": "N/A",
            "local-date": "2023-02-12",
            "local-time": "00:53:17.61",
            "max-apdu-length-accepted": 1024,
            "model-name": "N/A",
            "number-of-apdu-retries": 3,
            "object-identifier": "device,47241",
            "object-list": ["device,999", "network-port,1"],
            "object-name": "WebAI 240 Router",
            "object-type": "device",
            "property-list": [
                "object-identifier",
                "object-name",
                "object-type",
                "property-list",
                "system-status",
                "vendor-name",
                "vendor-identifier",
                "model-name",
                "firmware-revision",
                "application-software-version",
                "protocol-version",
                "protocol-revision",
                "protocol-services-supported",
                "protocol-object-types-supported",
                "object-list",
                "max-apdu-length-accepted",
                "segmentation-supported",
                "local-time",
                "local-date",
                "apdu-segment-timeout",
                "apdu-timeout",
                "number-of-apdu-retries",
                "device-address-binding",
                "database-revision",
                "active-cov-subscriptions"
            ],
            "protocol-object-types-supported": [],
            "protocol-revision": 22,
            "protocol-services-supported": [
                "acknowledge-alarm",
                "confirmed-cov-notification",
                "confirmed-event-notification",
                "subscribe-cov",
                "atomic-write-file",
                "add-list-element",
                "read-property",
                "read-property-multiple",
                "write-property",
                "write-property-multiple"
            ],
            "protocol-version": 1,
            "segmentation-supported": "segmented-both",
            "system-status": "operational",
            "vendor-identifier": 999,
            "vendor-name": "Sphere"
        },
        {
            "bacnet-ip-mode": "normal",
            "bacnet-ip-udp-port": 47808,
            "changes-pending": false,
            "ip-address": "192.158.2.241",
            "ip-subnet-mask": "255.255.255.0",
            "link-speed": 0.0,
            "mac-address": "192.158.2.241:47808",
            "network-number": 100,
            "network-number-quality": "configured",
            "network-type": "ipv4",
            "object-identifier": "network-port,1",
            "object-name": "NetworkPort-1",
            "object-type": "network-port",
            "out-of-service": false,
            "property-list": [
                "object-identifier",
                "object-name",
                "object-type",
                "property-list",
                "status-flags",
                "reliability",
                "out-of-service",
                "network-type",
                "protocol-level",
                "network-number",
                "network-number-quality",
                "changes-pending",
                "mac-address",
                "link-speed",
                "bacnet-ip-mode",
                "ip-address",
                "bacnet-ip-udp-port",
                "ip-subnet-mask"
            ],
            "protocol-level": "bacnet-application",
            "reliability": "no-fault-detected",
            "status-flags": []
        },
        {
            "bacnet-ip-mode": "normal",
            "bacnet-ip-udp-port": 47808,
            "changes-pending": false,
            "ip-address": "192.158.7.1",
            "ip-subnet-mask": "255.255.255.0",
            "link-speed": 0.0,
            "mac-address": "192.158.7.1:47808",
            "network-number": 200,
            "network-number-quality": "configured",
            "network-type": "ipv4",
            "object-identifier": "network-port,2",
            "object-name": "NetworkPort-2",
            "object-type": "network-port",
            "out-of-service": false,
            "property-list": [
                "object-identifier",
                "object-name",
                "object-type",
                "property-list",
                "status-flags",
                "reliability",
                "out-of-service",
                "network-type",
                "protocol-level",
                "network-number",
                "network-number-quality",
                "changes-pending",
                "mac-address",
                "link-speed",
                "bacnet-ip-mode",
                "ip-address",
                "bacnet-ip-udp-port",
                "ip-subnet-mask"
            ],
            "protocol-level": "bacnet-application",
            "reliability": "no-fault-detected",
            "status-flags": []
        }
    ]
}
`

JoelBender, 2024-05-17:
For this configuration you have two BACnet networks, `192.158.2.0/24` and `192.158.7.0/24`, the expectation is that any device on the first can reach `192.158.2.241:47808` with both unicast traffic to that address and broadcast address traffic sent to `192.158.2.255:47808`, similarly on the second network.

> The other devices (RPI) won't be on the same "network" even though they have the same ip range (192.168.7.1).

I can't quite picture how this would work because that's not an IP range, it's an IP address.  If the access point only provides this single IP address to a laptop, and the IP-to-IP router is using that same address (which doesn't make sense) then there is no way for it to send broadcast messages like Who-Is and have the router hear it.

An alternative configuration is to provide a BBMD at `192.158.2.241:47808` and have the laptops configure themselves as a foreign device that registers itself with the BBMD.  In this case the laptop will not be sending out broadcasts, but unicast messages to the BBMD saying "please forward this as a broadcast on my behalf" and the BBMD will forward broadcasts it receives as unicast messages to the laptop saying "I'm forwarding this broadcast to you on behalf of 192.158.2.201".  With respect to BACnet all of the devices are considered to be on the same network because they are all in the same broadcast domain, even if they are on separate IPv4 networks.

kheldaroz402, 2024-05-17:
let's see if I can explain it better.
We call our device a WebAI.
eth0 has two IP Address, eth0:1 and eth0:2
eth0:1 - runs a BACpypes application for  building controller (with AI's AO' etc) - this would be connected to the "bms" lan at the site
eth0:2 - is for a BACnet Router - this would be connected to the "bms" lan at the site
The WebAI also hosts an Wifi Access point, complete with DHCP server, NAT etc on wlan0 with a fixed IP of 192.168.7.1, so when a laptop connects it gets a DHCP address in the 192.168.7.1 range

Typically there would be one WebAI in mech board, so effectively each wifi network is unique as the range of it is not going to be large, particularly when its surrounded by metal

I hope that makes a little more sense

JoelBender, 2025-03-06:
Ok, looking at this again!  The BACnet router is between 192.168.3.41 on network 192.168.3.0/24 and 192.168.7.1 on the 192.168.7.0/24 network.  You need to assign BACnet network numbers to each of the IPv4 networks:

* 192.168.3.0/24 is network 100
* the first 192.168.7.0/24 is network 200
* the second 192.168.7.0/24 is network 300
* etc.

Now the router application on the first machine at 192.168.3.41 will be router from 100 to 200, the router at 192.168.3.43 will be from 100 to 300, etc.  When a PDU goes from the BACnet application at 192.168.3.40, it will be sent to 192.168.3.41  with a DNET/DLEN/DADR of `200:192.168.7.2` (using the BACpypes syntax to one of the Wifi addresses).  A PDU could also be sent to 192.168.3.43 with a destination `300:192.168.7.2` which is the same IPv4 Wifi address but through a different access point, so in BACnet terms it has a different BACnet address.