JoelBender/BACpypes3 #47: AttributeError: Sequence object has no attribute dict_contents

nveloso, 2024-05-28:
Hi!
When I'm trying to call `dict_contents()` on a `ReadPropertyMultipleACK` object I get the following error:

```
  File "...", line 392, in _dict_contents
    apdu_contents = apdu.dict_contents()
  File "...python3.8/site-packages/bacpypes3/apdu.py", line 583, in dict_contents
    self.apdu_contents(use_dict=use_dict, as_class=as_class)
  File "...python3.8/site-packages/bacpypes3/apdu.py", line 1037, in apdu_contents
    Sequence.dict_contents(self, use_dict=use_dict, as_class=as_class)
AttributeError: type object 'Sequence' has no attribute 'dict_contents'
```

It looks like `Sequence` does not have a `dict_contents` method implemented.

JoelBender, 2024-07-05:
The `dict_contents()` API is left over from "legacy" BACpypes and is a one-way mapping.  BACpypes3 has a `json` API that has `sequence_to_json()` and `json_to_sequence()`.  I would like to remove `dict_contents()` and the related functions like `apdu_contents()`.

JoelBender, 2025-03-06:
If this explanation is sufficient, please close the issue or continue the discussion.

nveloso, 2025-03-06:
Does that mean that the `apdu_contents` is deprecated? Right now it has a call to a function that does not exist.

If I have an instance of `ReadPropertyMultipleACK`, should I instead call `sequence_to_json` passing the `ReadPropertyMultipleACK` instance?

JoelBender, 2025-03-08:
Yes, `ReadPropertyMultipleACK` inherits from `Sequence` so you can get the sequence contents like this:
```
{
    "list-of-read-access-results": [
        {
            "object-identifier": "analog-input,1",
            "list-of-results": [
                {
                    "property-identifier": "description",
                    "read-result": {
                        "property-value": [
                            {
                                "opening": 4
                            },
                            {
                                "characterString": "00736e6f726b"
                            },
                            {
                                "closing": 4
                            }
                        ]
                    }
                },
                {
                    "property-identifier": "object-name",
                    "read-result": {
                        "property-access-error": {
                            "error-class": "property",
                            "error-code": "out-of-memory"
                        }
                    }
                }
            ]
        }
    ]
}
```
Note that `property-value` is a tag list, so the pieces inside it are tags and there could be all kinds of weirdness in that structure which is why it doesn't say simply `"snork"`.

nveloso, 2025-03-08:
Ok, thanks! I'll close this.

closed 2025-03-08 by nveloso