JoelBender/BACpypes3 #48: No response(s)

xkostrna, 2024-05-31:
# The problem
After restarting bacnet device we no longer recieve response to the **Who-Is service**. <br>

## Use-Case
1. We initialize the bacnet stack using this command:
`py -m bacpypes3 --address 192.168.251.3` <br>
2. We sent the **whois** command to the target ip **192.168.251.5** which is fully functional bacnet device like this:
`> whois 192.168.251.5`
3. At first we normally recieve the answer:
`5 192.168.251.5`
4. We then restart the bacnet connection of the device on IP **192.168.251.5**.
5. After restart we sent the **Who-Is** request again and we get No response(s) which is correct since the bacnet is loading:
`No response(s)`
6. After the device is successfully loaded we sent the **Who-Is** request again and the response is:
`No response(s)`

We are tracking everything with a Wireshark so we see when the device successfully connects to the bacnet. We also see the request and responses between our stack and the device. Everything works fine until we restart the bacnet on the target device. Then we can only see the `No response(s)` even when in the Wireshark everything seems fine. <br>

Evidence: <br>
[image]

JoelBender, 2024-06-01:
While the application at `192.168.251.5` is down the operating system is sending back the ICMP error, there is no application listening on the port.  When it is restarted it responds correctly again, but for some reason `192.168.251.3` isn't processing it.  Please run...
```
py -m bacpypes3 --address 192.168.251.3  --debug bacpypes3.ipv4
```
and see if you are getting something like this:
```
DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received b'\x81...
```
Which should be followed by something like this...
```

xkostrna, 2024-06-06:
Sorry for late response. <br>
Thanks for suggestion, I've run bacpypes3 as you suggested with argument `--debug`. Output of the `--debug` is working as you stated, I'm able to see PDU objects in both situations. Most important thing is that we figured out the behavior changes when I set my NetworkCategory in Windows. <br>

## Situation 1
I run this cmd: <br>
```
py -m bacpypes3 --address 192.168.251.3
```
I send `Who-Is` to the device on address 192.168.251.5 which is responding normally like this:
```
> whois 192.168.251.5
5 192.168.251.5
```
Now I restart the device so it doesn't respond which is OK and after restarting it responses again:
```
> whois 192.168.251.5
No response(s)
---After while----
> whois 192.168.251.5
5 192.168.251.5
```

## Situation 2
I set the NetworkCategory to private using this command:
```
Get-NetConnectionProfile -InterfaceAlias 'Ethernet 6' | Set-NetConnectionProfile -NetworkCategory Private
```
I run this cmd: <br>
```
py -m bacpypes3 --address 192.168.251.3
```
I send `Who-Is` to the device on address 192.168.251.5 which is responding normally like this:
```
> whois 192.168.251.5
5 192.168.251.5
```
Now I restart the device so it doesn't respond which is OK and after restarting i still can't see response in console however in Wireshark yes:
```
> whois 192.168.251.5
No response(s)
---After while----
> whois 192.168.251.5
No response(s) (even when Wireshark sees actual response)
```

JoelBender, 2025-03-06:
I don't have any way to reproduce this problem, hopefully you have found a solution.  If the debugging log contains the `DEBUG:bacpypes3.ipv4.IPv4DatagramProtocol:datagram_received` line saying the library received the packet but it's not making it up to the application layer, that would be a problem in the library.  If Wireshark sees it but the application doesn't, that's something else.

xkostrna, 2025-03-06:
Hi Joel,
I was consulting this thing over and over with our internal support, microsoft support etc, etc... one of the microsoft support guys sent me some diagnostic stuff, which was actually just bunch of scripts, suddenly problem disappeared. I talked about this with him and he told me to try `ipconifg /flushdns` command and guess what it worked. Nobody understands why.

jason-hongjae, 2026-03-16:
It was very useful for me.
ipconfig /flushdns

In my case, There was no respond for the wireshark at all from starting the app.

But after flushing dns. It works well.