JoelBender/BACpypes3 #49: Regarding the writemultiple support

BMPC2023, 2024-07-29:
Hi Sir,

@JoelBender 

Does BACpypes3 support the WriteMultiple property? If not, please guide me on how to implement it Sir.

bbartling, 2024-10-23:
Hey Joel, am I close at all? I am trying to mimic the `samples/read-property-multiple-console.py` but substitute for a write-multiple request.

```python
import sys
import asyncio
from typing import Callable

from bacpypes3.settings import settings
from bacpypes3.debugging import bacpypes_debugging, ModuleLogger
from bacpypes3.argparse import ArgumentParser
from bacpypes3.console import Console
from bacpypes3.cmd import Cmd
from bacpypes3.pdu import Address, IPv4Address
from bacpypes3.comm import bind
from bacpypes3.apdu import (
    ErrorRejectAbortNack,
    WritePropertyMultipleRequest,
    PropertyValue,
)
from bacpypes3.basetypes import PropertyReference, WriteAccessSpecification
from bacpypes3.primitivedata import ObjectIdentifier, Real, Unsigned
from bacpypes3.ipv4.app import NormalApplication
from bacpypes3.local.device import DeviceObject

# some debugging
_debug = 0
_log = ModuleLogger(globals())

# globals
app: NormalApplication = None

@bacpypes_debugging
class SampleCmd(Cmd):
    """
    Sample Cmd for WritePropertyMultiple
    """

    _debug: Callable[..., None]

    async def do_wpm(
        self,
        address: Address,
        *args: str,
    ) -> None:
        """
        usage: wpm address ( objid prop value priority )...
        """
        if _debug:
            SampleCmd._debug("do_wpm %r %r", address, args)

        # ensure proper argument structure
        if len(args) % 4 != 0:
            await self.response("Each property write requires: objid prop value priority")
            return

        args = list(args)
        parameter_list = []

        while args:
            # extract the object identifier, property identifier, value, and priority
            object_identifier = ObjectIdentifier(args.pop(0))
            property_identifier = args.pop(0)
            value = Real(float(args.pop(0)))  # assuming you're writing to Real values
            priority = Unsigned(int(args.pop(0)))

            # create PropertyReference for the write access
            property_reference = PropertyReference(
                propertyIdentifier=property_identifier
            )

[... 122 lines cut ...]
    await self.response(pdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/netservice.py", line 485, in confirmation
    await self.adapterSAP.process_npdu(self, npdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/netservice.py", line 1011, in process_npdu
    await self.response(pdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/comm.py", line 98, in response
    await self.serverPeer.confirmation(pdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 1669, in confirmation
    await tr.confirmation(apdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 593, in confirmation
    await self.await_confirmation(apdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 772, in await_confirmation
    await self.response(apdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 582, in response
    await self.ssmSAP.sap_response(apdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 1838, in sap_response
    await ServiceAccessPoint.sap_response(self, xpdu)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/comm.py", line 187, in sap_response
    await self.serviceElement.confirmation(*args)
  File "/home/bbartling/.local/lib/python3.10/site-packages/bacpypes3/app.py", line 946, in confirmation
    assert apdu.pduSource
AssertionError
```

JoelBender, 2024-10-24:
There isn't a `write_property_multiple()` method in `ReadWritePropertyMultipleServices` like there is for `read_property_multiple()` but it looks like @bbartling has a good start!  Review in process...

JoelBender, 2024-10-24:
@bbartling I put your code in [the sandbox](https://github.com/JoelBender/BACpypes3/blob/main/sandbox/write-property-multiple.py) and updated it a little (using `Application`).  I'm not exactly sure why your seeing that error, but it's coming from `do_WritePropertyMultipleRequest()` raising a `NotImplementedError`.

The code isn't quite up to snuff; the property array index isn't handled correctly, the priority is optional, and there can be more than one `PropertyValue` in the `listOfProperties`.  It also needs testing on something non-critical that supports the service!