JoelBender/BACpypes3 #50: Regarding UDP Communication Module

BMPC2023, 2024-07-31:
Hi Sir @JoelBender ,

I am currently working on BACpypes3. I need the UDPCommunication module for socket communication, which uses the UDPDirector and UDPActor. Please guide me on which classes I need to refer to in BACpypes3.

JoelBender, 2024-07-31:
Start with the samples in the [link layer](https://github.com/JoelBender/BACpypes3/tree/main/samples/link-layer) folder.  While UDP is a session-less peer-to-peer protocol, they are named things like "client" and "server" to make it easier to envision things like "initiate a service" and "respond to a request for a service".  Note that `UDPDirector` and `UDPActor` are terms from "legacy" BACpypes because at the time I thought that both the UDP and TCP portions of a stack should have the same design, not just the same request/indication/response/confirmation API.

Those samples use the lowest kind of [PDU](https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/pdu.py#L2016), you can of course make your own for your protocol and inherit from the generic [Client](https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/comm.py#L28) and [Server](https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/comm.py#L65).

JoelBender, 2025-03-06:
If this explanation is sufficient, please close the issue or continue the discussion.