JoelBender/BACpypes3 #52: Problem Receiving COV Notifications for Analog Outputs

et0614, 2024-08-08:
Hello,

I am grateful for the benefits of this very valuable project.

Please tell me about COV. 
I am creating code to register COV with the BACnet device on the 'DemoServer.exe' that comes with Yabe.

https://sourceforge.net/projects/yetanotherbacnetexplorer

I referred to the 'cov-client.py' file in the 'samples' directory and created the following code, but I am encountering an issue. 

async with self.bacdevice.change_of_value(
        address=Address("127.0.0.1:62980"),
        subscriber_process_identifier=1,
        monitored_object_identifier=ObjectIdentifier('analog-input:3333') # <- This works... but,
        # monitored_object_identifier=ObjectIdentifier('analog-output:3') # <- this doesn't
    ) as scm: #SubscriptionContextManager
        while True:
            property_identifier, property_value = await scm.get_value()
            print(f"{property_identifier} {property_value}")

While I am able to receive COV notifications properly for analog inputs, I am unable to receive them for analog outputs. 
I obtained the source code for Yabe's DemoServer and ran it in debug mode. 
It appears that COV registration via bacpypes is successful and COV notifications are being sent from the DemoServer. 
However, the bacpypes side is not receiving them.

I understand that COV functionality should be available for analog input, output, and value. Is this not correct? 
Or am I missing some configuration? 
Could you help me identify the cause of the issue?

JoelBender, 2024-08-11:
I can't think of anything that would cause the client to drop packets that it is receiving unless there is a mismatch in the context, which would be weird.  Time to dig in...  Please turn on debugging on the client by adding `--debug bacpypes3.service.cov` and look in the output log for the call to `do_ConfirmedCOVNotificationRequest()` which is where the COV notification is being received by the application.  There should be an entry in the `_cov_contexts` with a key that matches the address and subscriber process identifier that you provided in the creating of the context manager.

If it doesn't get this far, there's something "lower level" in the decoding that could be an issue.

et0614, 2024-08-12:
Thank you for your response.

Are you suggesting that I should pass "--debug bacpypes3.service.cov" as a command-line argument to my program? I tried that, but it seems that the logs are not being generated.

Would it be possible to write the following directly in the program?

```
import logging
....
logging.getLogger("bacpypes3.service.cov").setLevel(logging.DEBUG)
```

However, even with this approach, the logs are not being displayed. 
I primarily use C#, and I’m quite new to Python, so I apologize if I’m making any basic mistakes.

JoelBender, 2024-08-12:
Try this:
```
from bacpypes3.argparse import create_log_handler
...
create_log_handler("bacpypes3.service.cov")
```
In each module in the library there is a `_log` module level logger and a class or method level logger for each class or method in the module.  There is also a `_debug` flag in the module that is set when there is at least one log handler for any logger in that tree, but you'll notice that the `_debug` flag is checked before any of the debugging functions are called:
```
@bacpypes_debugging
class APCI(PCI):
    _debug: Callable[..., None]
    ...
    def __init__(self, *args, **kwargs):
        if _debug:
            APCI._debug("__init__ %r %r", args, kwargs)
```
Just the process of calling the `APCI._debug()` function adds quite a bit of overhead even if there are no log handlers for that specific logger (in this sample it's called "bacpypes3.apdu.APCI").  Checking for `if _debug:` is sufficiently fast that I can leave the debugging code in the library and when it's off it improves performance.

If you want to add a logger like you have done, just add the extra step of setting the flag:
```
import bacpypes3.service.cov
...
bacpypes3.service.cov._debug = 1
```
There are some other options like sending the output to a file or a rotating log handler outlined in the [documentation](https://bacpypes3.readthedocs.io/en/latest/gettingstarted/debugging.html).

et0614, 2024-08-13:
Dear Joel

Thank you for your detailed explanation. Thanks to your guidance, I was able to check the logs and realized where I went wrong.

I confirmed that I am receiving COV notifications for Analog Inputs, but not for Analog Outputs. Initially, I mentioned that both Analog Inputs and Analog Outputs were sending COV notifications from the Demo server. However, after further investigation, I discovered that COV notifications for Analog Outputs were not being sent.

Due to the fact that I stepped through the process line by line in debug mode, I mistakenly thought that the COV notification processing for a different request, which arrived with a delay, was a response related to Analog Outputs.

The issue remains as to why COV notifications are not sent for Analog Outputs, but I’ve confirmed that this is a matter related to the Demo server of YABE and has nothing to do with the calculations of BACpypes3.

I sincerely apologize for taking up your valuable time due to my misunderstanding. Once again, I deeply appreciate your very kind support.

Best regards

JoelBender, 2024-08-13:
No apology necessary!  I'm happy to help, and questions like these help other people as well as point me in the direction where more detailed documentation would be beneficial (which is, to be honest, just about everywhere :-)).