JoelBender/BACpypes3 #55: Migrating the Misty project to bacpypes3

BMPC2023, 2024-08-21:
Hi @JoelBender Sir,

Below are the challenges I am facing sir

misty project => [link](https://github.com/riptideio/misty)

1)  The Misty project primarily uses UNIX domain sockets (UDS) for communication between the Python components and the BACnet C stack. I am wondering if it's possible to implement a similar UDS-based communication mechanism within BACpypes3. Are there any existing provisions or best practices for this in BACpypes3?

2)  I have identified that the IPv4DatagramServer class in BACpypes3 is utilized for asynchronous UDP socket communication. I am considering adapting this class to bind a UDS socket for communication with the mstp_agent.c file in the Misty project, which is responsible for RS485 communication. Would you suggest this approach, or is there a more suitable method within BACpypes3?

The above points are identified by comparing the MSTPSimpleApplication(which is modified from BIPSimpleApplication of bacpypes stack) and NormalApplication (bacpypes3 stack)

JoelBender, 2024-08-27:
I'm expecting to be able to use a Network Port Object for configuration information, the properties would match the parameters you give to the agent:

* **Network_Interface_Name** (Clause 12.56.20) matches the "interface"
* `MAC_Address` (Clause 12.56.15) matches the "address"
* `Link_Speed` (Clause 12.56.17) matches the "baudrate"
* `Max_Master` (Clause 12.56.51) matches the "max masters"
* `Max_Info_Frames` (Clause 12.56.52) matches the "max info frames"

A JSON configuration would look like this:
```json
        {
            "interface": "/dev/ttyS0",
            "link-speed": 38400.0,
            "mac-address": "10",
            "max-info-frames": 1,
            "max-master": 127,
            "network-number": 100,
            "network-number-quality": "configured",
            "network-type": "mstp",
            "object-identifier": "network-port,1",
            "object-name": "NetworkPort-1",
            "object-type": "network-port",
            "out-of-service": false,
            "protocol-level": "bacnet-application",
            "reliability": "no-fault-detected"
        }
```

JoelBender, 2024-08-28:
1) Are you expecting that there will be (a) one agent that communicates with multiple serial ports or (b) one agent per serial port?  It looks like the agent can be configured with multiple serial ports (the `port_info_array`) so somehow the specific port (the `port_index`?) for application/agent PDUs needs to be encoded, not just the MSTP data.

2) What MSTP frames are passed between the application an agent?  I am expecting to just see types 5, 6, 32, 33, and 128 through 255 so the agent will take care of the polling and 7 (reply postponed).

JoelBender, 2024-08-28:
The branch contains a starting point.  Run the agent in one terminal...
```shell
$ python3 samples/misty-agent.py /tmp/tmp.DdsXSDW5pv --debug
```
...and the application in another...
```shell
$ python3 samples/misty-application.py --json samples/misty-application.json --debug bacpypes3.mstp --color
```
Note that the `network-interface-name` in the JSON configuration file is passed to the agent.  The first octet for downstream PDUs (from the application to the agent) is the destination, the first octet is the source for upstream PDUs.

The agent simply echos the PDUs, so a read request from the application is sent to the agent and bounces back, likewise the data is bounced back which is why this works:
```
> read 1 device,997 object-name
Potemkin-0
```

shashankgowdasd, 2024-08-28:
Hi @JoelBender Sir,

Thank you for the work you’ve done on the samples. I went through them and successfully tried them out. I have a few questions:

The original Misty project utilized the "BACnet C stack" for a full-fledged MSTP application. Similarly, is it possible to link the mstp_agent.py to the C stack for full MSTP support? If so, could you provide guidance on how best to achieve this integration?

BMPC2023, 2024-08-28:
Hi @JoelBender Sir,

Thank you for the work you’ve done on the samples. I went through them and successfully tried them out. I have a few questions:

The original Misty project utilized the "BACnet C stack" for a full-fledged MSTP application. Similarly, is it possible to link the mstp_agent.py to the C stack for full MSTP support? If so, could you provide guidance on how best to achieve this integration?

JoelBender, 2024-08-28:
I'm not sure what you mean by "full MSTP support."

I have successfully use [pybind11](https://github.com/pybind/pybind11) to call C++ libraries from Python, but it was just a library of functions and nothing that would require threads or bi-directional communications.  I was hoping that the agent would be a C application that would take care of everything except the "data" PDUs like those in (2) above.  Maintaining the MSTP state machine and token passing in Python would probably be a bad idea for performance.

Without any experience with the library, I would start with the [MSTP router](https://github.com/bacnet-stack/bacnet-stack/tree/master/apps/router-mstp) sample, keep the MSTP layer, and chop off the network layer an IPv4 stuff.  Packets that would normally be received from the network layer but are now being received from the socket would eventually call [dlmstp_send_pdu](https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/router-mstp/main.c#L338) and keep the [dlmstp_receive](https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/router-mstp/main.c#L1195) loop.

I get the nagging feeling that you're already doing that!

JoelBender, 2024-08-28:
In this mindset, the agent essentially becomes an RPC API server where the messages going into and out of the agent are its own protocol and it would be fairly simple to define the message format as a C structure that maps to a [dataclass](https://docs.python.org/3/library/dataclasses.html).  Are you thinking like having [get_mstpstats](https://github.com/riptideio/misty/blob/master/misty/mstplib/mstp_agent.c#L154) available as a BACpypes function?

BMPC2023, 2024-08-29:
> I'm not sure what you mean by "full MSTP support."

Full MSTP support means how the Misty project integrates with the BACnet C stack to provide all MSTP token-passing and other features. Can BACpypes3 be made to support these features as well?

> I have successfully use [pybind11](https://github.com/pybind/pybind11) to call C++ libraries from Python, but it was just a library of functions and nothing that would require threads or [...]

Yes, Sir. I was attempting to link mstp_agent.c with the BACnet C stack, as referenced. However, I’m encountering challenges in figuring out how to pass the PDU packets to dlmstp_send_pdu using a single socket path.

> In this mindset, the agent essentially becomes an RPC API server where the messages going into and out of the agent are its own protocol and it would be fairly simple to define the message format as [...]

The get_mstpstats function, used in the earlier Misty code, is for checking MSTP network-related parameters such as tokens and communication statistics. Since it’s not directly related to the core functionality we’re focusing on, we can ignore that function, Sir.

JoelBender, 2024-09-01:
> I was attempting to link mstp_agent.c with the BACnet C stack, as referenced. However, I’m encountering challenges in figuring out how to pass the PDU packets to dlmstp_send_pdu using a single socket path.

I suspect this is a blocking IO issue and I dug around looking for `select()` and other low level IO calls and didn't find an answer. Sending a ping to @skarg to see if I'm going down the right path.

skarg, 2024-09-04:
Usually the datalink for MSTP on an OS runs as a POSIX thread or as Win32 Thread and interfaces into an rs485 module downstream and presents a datalink API upstream. I haven't looked at misty in awhile, but I think it presents the python to c bindings as an agent, and modified the ports/linux/dlmstp_linux.c datalink to support Linux and Darwin in the BACnet Stack C library.

The ports/linux/dlmstp_linux.c module includes a variety of OS features to run the transmit/ receive thread dlmstp_master_fsm_task() and keep the data protected with semaphore from a calling thread in a shared data structure.

Not sure if that helps?

BMPC2023, 2024-09-07:
Hi @JoelBender Sir,
I wanted to check if there have been any updates regarding the integration of the BACpypes3 API with the BACnet C stack.