JoelBender/BACpypes3 #56: Commandable Objects

kheldaroz402, 2024-08-30:
Hi Joel,

I've started converting my code to bacpypes3.

In bacpypes we had commandable objects - do they exist in bacpypes3 or is it done differently?

eg 
```
from bacpypes.local.object import (
    AnalogValueCmdObject,
    AnalogOutputCmdObject,
    BinaryOutputCmdObject,
    BinaryValueCmdObject,
    DateValueCmdObject,
    MultiStateValueCmdObject,
    MultiStateOutputCmdObject

)
```

bbartling, 2024-08-30:
See this link to a gist. Legacy bacpypes repo has a mini device in the
samples for a simple BACnet server and there is a bacpypes3 version but I
don't think its in the repo. You have to make a Class for a commandable
object in your app instead importing a commandable object which was the
legacy bacpypes way.

mini_device_revisited.py

https://gist.github.com/bbartling/573030a2facc97e178b2362606b9cdda#file-mini_device_revisited-py

Let me know if you have any questions hopefully the link works I am typing
from my phone.

JoelBender, 2024-09-01:
Look in the `bacpypes3.local...` modules which are intended for servers/gateways:
```python
from bacpypes3.local.binary import BinaryValueObjectCmd
from bacpypes3.local.analog import AnalogValueObjectCmd
```
Note that `BinaryOutputObject` and `AnalogOutputObject` are always `Commandable` and there are other variants as well, like intrinsic reporting and fault detection.  There are _lots_ of places for improvements, _especially_ considering the overlap of different kinds of functionality and side effects (by which I mean changes in one property imply changes to other properties).

All of these things are going to need tests as outlined by [ASHRAE 135.1](https://webstore.ansi.org/standards/ashrae/ansiashraestandard1352019).

p.s.- I didn't realize I swapped the `Object` and `Cmd` part.  Humph.

JoelBender, 2025-03-06:
If this explanation is sufficient, please close the issue or continue the discussion.

kheldaroz402, 2025-06-05:
no it won't let me

I managed to get it working not sure if its the right way but it does work

`#!/opt/WebAI/WebAIPy/virtualenv/bin/python3

# ──────────────────────────────────────────────────────────────────────────
#  Flip DEBUG = True if you want full bacpypes3 logging to trace behavior.
# ──────────────────────────────────────────────────────────────────────────
DEBUG = False
_debug = DEBUG

import os
import sys
import types
import asyncio
import logging
from datetime import datetime

import ifaddr

from config.settings   import poolSettings
from src.utils         import (
    get_local_ip,
    get_subnet_mask,
    get_subnet_cidr,
    calculate_broadcast_address,
    color,
)
from src.redis_handler import RedisHandler
from src.db_batcher    import DBBatcher

# ──────────────────────────────────────────────────────────────────────────
#  BACpypes 3 imports
# ──────────────────────────────────────────────────────────────────────────
from bacpypes3.app               import Application
from bacpypes3.constructeddata   import ArrayOf
from bacpypes3.basetypes         import PriorityValue, TimeValue, DateRange, DailySchedule, BinaryPV, CharacterString
from bacpypes3.local.device      import DeviceObject
from bacpypes3.local.networkport import NetworkPortObject
from bacpypes3.local.analog      import AnalogValueObject, AnalogOutputObject, AnalogInputObject
from bacpypes3.local.binary      import BinaryValueObject, BinaryOutputObject, BinaryInputObject
from bacpypes3.local.multistate  import MultiStateValueObject, MultiStateOutputObject, MultiStateInputObject
from bacpypes3.local.cmd         import Commandable
from bacpypes3.local.schedule    import ScheduleObject
from bacpypes3.primitivedata     import Time, Date, Boolean, Integer, Null, Real, Unsigned, Enumerated
from bacpypes3.debugging         import bacpypes_debugging, ModuleLogger

# ──────────────────────────────────────────────────────────────────────────
#  Configure logging: your own logger = DEBUG; BACpypes 3 = WARNING
# ──────────────────────────────────────────────────────────────────────────
logging.basicConfig(level=logging.DEBUG)
_log = logging.getLogger(__name__)
_log.setLevel(logging.DEBUG)
# Silence BACpypes 3 internals unless they hit WARNING or ERROR
logging.getLogger("bacpypes3").setLevel(logging.WARNING)

from bacpypes3.argparse import create_log_handler
if DEBUG:
    create_log_handler("bacpypes3.json.util");    import bacpypes3.json.util;    bacpypes3.json.util._debug = 1
    create_log_handler("bacpypes3.app");          import bacpypes3.app;          bacpypes3.app._debug = 1
    create_log_handler("bacpypes3.service.cov");  import bacpypes3.service.cov;  bacpypes3.service.cov._debug = 1
    create_log_handler("bacpypes3.apdu");         import bacpypes3.apdu;         bacpypes3.apdu._debug = 1
    create_log_handler("bacpypes3.ipv4");         import bacpypes3.ipv4;         bacpypes3.ipv4._debug = 1
    create_log_handler("bacpypes3.local.cmd")
    bacpypes3.local.cmd._debug = 1

_log = ModuleLogger(globals())

# ──────────────────────────────────────────────────────────────────────────
[... 509 lines cut ...]
        mso1,  # <<— MSO1 instead of MSV2
        ai1, bi1, msi1
    ]

    app = Application.from_object_list(all_objects)
    logging.info(">>> Sending I-Am on startup…")
    app.i_am()

    redis   = RedisHandler(cfg.get("redis_url") or os.getenv("Redis_URL"))
    await redis.connect()
    batcher = DBBatcher(max_batch=int(cfg.get("db_max_batch", 50)))
    app.redis   = redis
    app.batcher = batcher

    # Never exit, so that I-Am remains on the network
    await asyncio.Future()

# ──────────────────────────────────────────────────────────────────────────
if __name__ == "__main__":
    try:
        asyncio.run(main())
    except KeyboardInterrupt:
        _log.info(f"{color.HOT_PINK}Shutting down BACnet application.{color.END}")
        sys.exit(0)
`