JoelBender/BACpypes3 #57: Invalid state transition from COMPLETED to AWAIT_CONFIRMATION

JoelBender, 2024-09-02:
Every once in a while this exception is raised.  Upon investigation, `indication()` is being called from `await_confirmation_timeout()` [here](https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/appservice.py#L852) as if the request is coming from the application but that's not actually true.  The state is COMPLETED, so that means that the response was received but the timeout task wasn't canceled.  The `Task` object that returned by [ensure_future()](https://github.com/JoelBender/BACpypes3/blob/main/bacpypes3/appservice.py#L623) should be canceled, or better associated with the `_timer_handle()`.

@ChristianTremblay

ChristianTremblay, 2024-09-07:
```
1
2
3
4
5
6
7
8
9
10
11
12
Task exception was never retrieved
future: <Task finished name='Task-38353' coro=<ClientSSM.await_confirmation_timeout() done, defined at /home/ctremblay/.local/lib/python3.10/site-packages/bacpypes3/appservice.py:836> exception=RuntimeError('invalid state transition from COMPLETED to AWAIT_CONFIRMATION')>
Traceback (most recent call last):
  File "/home/ctremblay/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 852, in await_confirmation_timeout
    await self.indication(self.segmentAPDU)
  File "/home/ctremblay/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 536, in indication
    self.set_state(AWAIT_CONFIRMATION, self.apduTimeout)
  File "/home/ctremblay/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 404, in set_state
    SSM.set_state(self, newState, timer)
  File "/home/ctremblay/.local/lib/python3.10/site-packages/bacpypes3/appservice.py", line 216, in set_state
    raise err
RuntimeError: invalid state transition from COMPLETED to AWAIT_CONFIRMATION
```
that was on Gitter but it's probably better to put that here as reference

This is running on a Raspberry Pi, Ubuntu 64bit

ohshazbot, 2025-08-13:
I put up a PR for this at https://github.com/JoelBender/BACpypes3/pull/96