JoelBender/BACpypes3 #60: sc connect example using ssl context to secure the connection with tls

reider-dev, 2024-10-04:
Would it be possible to adapt the link-layer sc echo server and client examples to include an ssl context such that mutual authentication is enforced? 

I have both examples running in independent docker containers and can see the echo functioning well.  However, all my attempts to simply adapt the examples (and/or the bacpypes3.sc.service file) have failed to show the same functionality when trying to connect to device with wss://uri.

Any feedback/thoughts would be greatly appreciated. My feeling is that it should be trivial, and I am missing something simple.

Another question: Is it expected that the above mentioned echo server doesn't work in anything greater than python 3.10? I tried bumping the version in the (my) Dockerfiles, but then the basic echo functionality stopped working.

JoelBender, 2024-10-07:
There is a [new API](https://websockets.readthedocs.io/en/stable/howto/upgrade.html#upgrade-to-the-new-asyncio-implementation) for the websockets library, thank you for the issue.  Apparently you can pass additional arguments on the server side that get passed to [create_server()](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.create_server), similarly the client side there is [create_connection()](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.create_connection).  There are some interesting tidbits [here](https://stackoverflow.com/a/57154483) for mutual authentication.  I can fix the API piece, but getting the tooling correct for generating certificates is still a mountain I haven't climbed (cliff maybe? every time I start it looks more like a wall).

reider-dev, 2024-10-07:
Thanks for the response and information. If the API piece is working, I image the tooling to generate certificates is anyway out of scope. Most use cases would expect the user to bring their own.  At the moment I am trying to simulate one or more trivial BACnet enabled devices, as docker containers, securely connecting to a BACnet hub (a separate docker container). For the moment I simply generated my own authority and signed certificates with openssl.  Thanks again!