JoelBender/BACpypes3 #61: Instance Numbers

kheldaroz402, 2024-10-22:
i was just reading

```
Q.3.11.18 <ObjectIdentifierPattern>
BACnet Object Identifier primitive data that allows independent specification of type and instance is encoded with the
element <ObjectIdentifierPattern>. In addition to the common attributes and child elements described in Clauses Q.3.1 and
Q.3.2, the <ObjectIdentifierPattern> element can also have the value specifiers described in Clause Q.3.5, the range
restrictions described in Clause Q.3.6, and the named values described in Clause Q.3.3.
The value of an <ObjectIdentifierPattern> elements is an xs:string. The format of this string is "T,N", where
T is either a type identifier or a single asterisk ("*") character to indicate "unspecified", The type identifier is either a decimal
number with no leading zeroes, or a standard type name exactly equal to the names specified in the definition for
BACnetObjectTypes in Clause 21, or from the XML definition of "0-BACnetObjectType", if available.
N is either an instance number or a single asterisk ("*") character to indicate "unspecified". The instance number is a decimal
number with no leading zeroes.
When the type is "unspecified", the value encoded for BACnet binary communications is a local matter. When the instance
number is "unspecified", it shall be encoded for BACnet binary communications as the value 4194303
```

If the instance number is a decimal - that implies you can have an instance of -10 or 9.5 or 0 - seems a bit strange to me

JoelBender, 2024-10-23:
In this case "decimal" simply means a number using base 10, "unsigned integer" would be more appropriate.  If you would like to submit an editorial change to the standard that would be considered by the SSPC.  By being an instance number it is already an unsigned integer.

kheldaroz402, 2024-10-23:
So that would mean zero is a legitimate instance number - which seems a bit inconsistent since most BACnet arrays start at 1
but thanks for the clarification :)

closed 2024-10-23 by kheldaroz402

JoelBender, 2024-10-23:
Yes, zero is a legitimate instance number and there is a major vendor that predefines Analog Value and Binary Value Objects 0..99, even if they aren't used in the control sequence(s) in the device.  It's not an "index", it's just a number, and there can be gaps, for example when a program/sequence is deleted and replaced with something else the old objects go away.