JoelBender/BACpypes3 #63: Changing of Properties fails to trigger EventAlgorithm (Intrinsic Alarming)

juraj3579, 2024-11-27:
Hello,
first of all, thank you very much for your work on bacpypes3. Amazing work.

I stumbled upon problem, when using Intrinsic Alarming with bacpypes3. 
I hope I can explain the issue somehow understandable.

I'm using your example event-server-avo1-ir.py with some modifications (json config)

Got in problems, when trying to change properties (lowLimit, highLimit, limitEnable), which should trigger the EventAlgorithm.
When I change  presentValue, the Algorithm works as intended. Changing other properties doesn't trigger the state transitions. 
Important to mention, this issue happens only, when I'm changing the properties using other client (BACEye). When setting the properties over the command line all works fine.

Tried to debug the code, not sure, if in right place. 
But I can see, the properties are all bind to the properties_monitors of the object and got the property_change method assigned. However this method and the __setattr__ of the Object class gets called only when changing present value property using BACEye. Changing other properties doesn't trigger this method, however the values of properties are accepted.  

Did some test with highLimitEnable starting from Normal State:
Other Client set to 1 -> no change of state
CommandLine set to 1 -> no change, since it's already 1
Other Client set to 0 -> no change of state
CommandLine set to 1 -> change of state to highLimit

Other Client set to 1 -> no change of state
CommandLine set to 1 -> no change, since it's already 1
CommandLine set to 0 -> no change of state
CommandLine set to 1 -> change of state to highLimit

Is there a difference when writing properties using other BACnet client and command line except for presentValue? Or is there some other issue?

JoelBender, 2024-11-29:
You are correct, this is an early design choice that I made.  The intent is to separate the "internal" and "external" reads and writes to the property of an object, so that `valu = obj.prop` and `obj.prop = valu` are considered "internal" operations while `obj.read_property()` and `obj.write_property()` are called from a client request (like "external" to the application).  Operations that are "internal" are left as simple as I could get them and still have the "magic" of keeping content correct according to the property type. (see sidebar 1)

The assumption is that a developer knows the value has changed and might be making other kinds of changes (like changing both the high limit and low limit) before expecting "side effects" like triggering notifications.  There are lots of property value assignments that are made, particularly during initialization, that are meaningless by themselves and must all be completed before other things happen.  The other consideration is that "internal" changes are not awaitable (something like `await obj.prop = valu`) but `read_property()` should be a co-routine so that library users can override the implementation of property values which might in turn use async functions.

----

### sidebar 1
```python
>>> from bacpypes3.primitivedata import Real
>>> from bacpypes3.constructeddata import Sequence
>>> class MyThing(Sequence):
...     myProperty: Real
... 
>>> x = MyThing(myProperty=4)
>>> x.debug_contents()
    myProperty = 4.0
>>> 
```
I thought this kind of "type casting" is less surprising for Integer/Unsigned, and Real/Double types, but other ones might seem like it's pushing the "do what I mean, not what I say" boundary:
```
>>> from bacpypes3.basetypes import BinaryPV
>>> class AnotherThing(Sequence):
...     myProperty: BinaryPV
... 
>>> x = AnotherThing(myProperty=True)
>>> x.debug_contents()
    myProperty = <BinaryPV: active>
```
The tradeoff is that when the property value is accessed via `read_property()` the value is already the correct type, and other types of side effects can be enforced:
```
>>> from bacpypes3.constructeddata import Choice
>>> class Snork(Choice):
...     pReal: Real
...     pBinaryPV: BinaryPV
... 
>>> s = Snork(pReal=12)
>>> s.debug_contents()
    pReal = 12.0

>>> s.pBinaryPV = "inactive"
>>> s.debug_contents()
    pBinaryPV = <BinaryPV: inactive>
```

JoelBender, 2025-03-06:
If this explanation is sufficient, please close the issue or continue the discussion.