JoelBender/BACpypes3 #66: Unable to read foreign properties

desultory, 2025-02-08:
I'm trying to make a simple program which can pull data from devices on another network and export those stats in another format.

Using this code, I get the error `bacpypes3.apdu.Error: communication: unknown-route`:

```
#!/usr/bin/python3

from asyncio import run

from bacpypes3.ipv4.app import ForeignApplication
from bacpypes3.local.device import DeviceObject
from bacpypes3.pdu import Address
from bacpypes3.primitivedata import ObjectIdentifier
from bacpypes3.settings import settings

class BACNetExporter(ForeignApplication):
    def __init__(self, *args, **kwargs):
        this_device = DeviceObject(
            modelName="BACNetExporter",
            objectName="BACNetExporter",
            objectIdentifier=1234,
        )
        super().__init__(this_device, Address("192.168.254.12"), *args, **kwargs)

async def main():
    settings["route_aware"] = True
    exporter = BACNetExporter()
    address = Address("9000:10.1.8.10")
    object_id = ObjectIdentifier("binaryOutput:6659")
    print(await exporter.read_property(address, object_id, "presentValue"))

if __name__ == "__main__":
    run(main())
```
I'm able to communicate with this device using YABE.

trying read_property, I see it attempts to register the foreign device and doesn't just send requests like YABE does:

`python read-property.py 9000:10.1.8.10 binaryOutput:1 presentValue --address 192.168.254.12 --foreign 10.1.8.10`

JoelBender, 2025-02-09:
Yes, the `ForeignApplication` expects to be able to register as a foreign device, if it can't then broadcast services like Who-Is-Router-To-Network won't work.  This code looks like it is running on network X (which is unknown) and reading from a device on network 9000, and registering as a foreign device on that same network.

If you are **not** on network 9000 and the device is on network 9000 then there should be an IPv4-to-IP4 router sitting in the middle.  Assuming that device is at 10.20.30.40, the "route aware" destination for the request would be "9000:10.1.8.10@10.20.30.40" which means "send this packet to 10.20.30.40 to be forwarded to the device at 10.1.8.10 on network 9000".  That will prevent BACpypes from issuing a Who-Is-Router-To-Network.

If the BBMD at 10.1.8.10 is on network 9000 and you can talk to it directly, you don't have to register as a foreign device (or use the ForeignApplication) just send the request directly to 10.1.8.10.

If YABE is working, please provide a hex dump or decoding of the UDP packets being sent and received and I can tell you what your topology looks like from the combinations of SNET/SLEN/SADR and DNET/DLEN/DADR fields of the NPCI.  Then it will be simple to set up your application.  It would also be informative if you can see that YABE is registering as a foreign device _before_ the read request is sent so that it already has the IPv4 address of the router to network 9000 in its cache.

desultory, 2025-02-09:
A packet from wireshark as a hex stream:

4500002de8b700004011c148c0a8fe0c0a01080ab686bac00019d0ea810a001101040275000c0c02002328194c

I may have been using the wrong terms, sorry. I connect to the device using "9000-10.1.8.10:47808" in yabe, there are no other BACNet devices involved, so I think direct communication is possible.

Are there any docs on what is needed to make an application? I was originally trying to do something similar to the  from_args static method, but looking through the bugs, I think the NormalApplication may make sense to use as a base class here.

Thanks for the help!

JoelBender, 2025-02-09:
> I think I should be trying to talk to "device 9000" and since I'm on a different network, I want to talk to the device directly (this is over wireguard so broadcasting isn't much of an option)

Ah ha!  Now I get it, I was over interpreting the 9000.  That's the device identifier (also known as the device instance number because it's the instance number of the device object, e.g., "device,9000").  For your sample...

Starting at the BVLL, which contains BVLCI header on the front, `81.0a.00.11`; this is an original unicast message from one device to another, no fancy "by the way this came from somebody else" or "please broadcast this for me" stuff.

The next layer is the NPDU, which contains an NPCI header on the front, `01.04`, which is also really simple; this is a message from one device to another on the same BACnet network, no fancy "please forward this someplace because you are a router" (missing DNET/DLEN/DADR) or "I'm forwarding this to you because I'm a router" (missing SNET/SLEN/SADR).

The next layer is the APDU and that's the Read-Property request.

For this same behavior in BACpypes3, try running this:

```
$ python3 -m bacpypes3 --address 192.168.254.12
> read 10.1.8.10 binary-output,6659 present-value
...
> read 10.1.8.10 device,9000 object-name
...
```
> Are there any docs on what is needed to make an application? I was originally trying to do something similar to the from_args static method, but looking through the bugs, I think the NormalApplication may make sense to use as a base class here.

The sample applications are built up from the `start_here.py` sample, the intent is to add bits and pieces of various types of functionality.  The advantage of using `Application.from_something(...)` is that the same code can run with CLI parameters, environment variables, or a JSON configuration file where it is directly on a BACnet network or as a foreign device.  In this case, you are correct that a `NormalApplication` is good enough.

The `read_property.py` sample application would look like this:

```
$ python read_property.py 10.1.8.10 binary-output,6659 present-value --address 192.168.254.12
...
```

desultory, 2025-02-10:
Thanks for that info, I may try to help improve docs/examples. I'm only mildly familiar with BACnet and was using this to learn more.

Is there a clear description of the expected formats for args? Is this a common BACnet syntax? I mean like network:address:port? YABE was using instanceid-ip:port

In that example, I don't see how I should set the device ID in read_property.

I tried running: 
```
python3 -m bacpypes3 --address 192.168.254.12
> read 10.1.8.10 binary-output,6659 present-value
```
it hangs, but I do not se any network activity.

desultory, 2025-02-10:
Ah my bad, I had YABE running and it was binding to the interface already, the REPL didn't let me know until I killed it.

How is it able to know instance id 9000 is in use when that first command is used?

desultory, 2025-02-10:
At this point I'm still a bit confused how I'm supposed to make a NormalApplication function properly. It seems to be trying to broadcast and fails because it's on a wireguard interface.

desultory, 2025-02-10:
I've got this working, is there any reason the address fields can't be cast to an Address type from strings automatically?

```
#!/usr/bin/python3

from asyncio import run
from zenlib.logging import loggify

from bacpypes3.ipv4.app import NormalApplication
from bacpypes3.local.device import DeviceObject
from bacpypes3.pdu import Address
from bacpypes3.apdu import AbortPDU
from bacpypes3.primitivedata import ObjectIdentifier

def force_to_address(func):
    def _to_address(address):
        if not isinstance(address, Address):
            address = Address(address)
        return address

    def wrapper(self, *args, **kwargs):
        if address := kwargs.pop("address", None):
            address = _to_address(address)
            kwargs["address"] = address
        else:
            args = list(args)
            args[0] = _to_address(args[0])
            args = tuple(args)
        return func(self, *args, **kwargs)

    return wrapper

def get_oid(oid_name, instance_id=1):
    return ObjectIdentifier(f"{oid_name},{instance_id}")

@loggify
class BACNetExporter(NormalApplication):
    @force_to_address
    def __init__(self, address, *args, **kwargs):
        this_device = DeviceObject(
            modelName="BACNetExporter",
            objectName="BACNetExporter",
            objectIdentifier=1234,
            segmentationSupported='segmentedBoth',
        )
        super().__init__(this_device, address, *args, **kwargs)

    @force_to_address
    async def get_device_name(self, address, instance_id=1):
        oid = get_oid("device", instance_id)
        return await self.read_property(address, oid, "objectName")

    @force_to_address
    async def get_object_list(self, address, instance_id=1):
        oid = get_oid("device", instance_id)
        try:
            return await self.read_property(address, oid, "object-list")
        except AbortPDU as e:
            if str(e.apduAbortRejectReason) != "segmentation-not-supported":
                self.logger.error(f"Error reading object list: {e}")
                raise e

        object_list = []
        list_size = await self.get_object_list_item(address, instance_id)
        for i in range(1, list_size):
            object_info = await self.get_object_list_item(address, instance_id, i)
            self.logger.warning(f"Object info: {object_info}")
            object_list.append(object_info)
        return object_list

    @force_to_address
    async def get_object_list_item(self, address, instance_id=1, item_id=0):
        oid = get_oid("device", instance_id)
        return await self.read_property(address, oid, "object-list", item_id)

async def main():
    exporter = BACNetExporter("192.168.254.12")
    #object_id = ObjectIdentifier("binaryOutput:6659")
    print(await exporter.get_device_name("10.1.8.10", 9000))
    print(await exporter.get_object_list("10.1.8.10"))
    #print(await exporter.read_property("10.1.8.10", object_id, "presentValue"))

if __name__ == "__main__":
    run(main())
```

JoelBender, 2025-02-12:
For the current version 0.0.99 (or later) the `read_property()` can take strings for the parameters, the property reference can be something like `priority-array[6]`.

```python
    async def read_property(
        self,
        address: Union[Address, str],
        objid: Union[ObjectIdentifier, str],
        prop: Union[PropertyIdentifier, str],
        array_index: Optional[int] = None,
    ) -> _Any:
        ...
```

Taking strings is a relatively new feature, I'll update the sample applications and it looks like the [documentation](https://bacpypes3.readthedocs.io/en/latest/samples/read-property.html) hasn't updated itself automatically :-).

desultory, 2025-02-12:
Ah, I'm on bacpypes3 0.0.98, I was about to consider making a PR, I'm glad this is becoming a builtin feature, I can remove my decorator :D

bbartling, 2025-02-12:
I've been following this thread... Just for the newbie you are just
attempting to read data from a remote sub network right via BACnet
requesting? Which stemmed this git issue... When I run into this scenario
what do people do?!

desultory, 2025-02-12:
> I've been following this thread... Just for the newbie you are just
> attempting to read data from a remote sub network right via BACnet
> requesting? Which stemmed this git issue... When I run into this scenario
> what do people do?!
> […](#)

I was attempting to read from a remote device, and was mistakenly specifying the device ID as the virtual/sub network. In the past I used BACNet on the same network as devices and thought the device ID must be passed, when it seems implied when not broadcasting.

JoelBender, 2025-02-12:
The latest update [here](https://github.com/JoelBender/BACpypes3/commit/d31b3655ea10cc5598dfaef3f3b5b123cd6b5b5c) touches the sample applications but does not touch the `__main__.py` functions.  Let's turn that into a [discussion](https://github.com/JoelBender/BACpypes3/discussions/67).

closed 2025-02-12 by JoelBender

bbartling, 2025-02-12:
Thanks @desultory for the insight. In Building Automation contracting at least in my experience it is very rare for a BAS technician person to use BACnet to read data from a remote network where at least the product line I setup we would just use a supervisory level device to integrate from one network to another...but ever since I graduated into the IoT world with some new skills it almost seems this is quite common. 

But I did actually witness recently in a campus of buildings a flat IT network for OT and each building was on its own port number with a BBMD, so IDK each contractor and manufacture may do things differently. I think I noticed on your git profile you do security research...feel free to ping me on anything the BAS people do in practice in the wild, I have a lot of field experience setting up BAS.