JoelBender/BACpypes3 #71: Using write-property-console.py to send a DateTime value

cadam01, 2025-03-07:
Hello,

I am using write-property-console.py to do some testing on a modified DateTimeValue object I am working on.  I am having trouble sending a DateTime value.  I am not getting past the encoding of the DateTime value on the write-property-console.py side.  The proper format of DateTime value at the console line has eluded me.   

I launch it with:

$ python write-property-console.py --debug bacpypes3.constructeddata

One of my write attempts:

> write 192.168.1.124 datetimeValue:1 presentValue date=(125,6,1,1),time=(1,1,1,1)

The resulting error is: 

write error: date is a required element of DateTime | Error type : <class 'AttributeError'>

Below is the debug output.

Chris

> write 192.168.1.124 datetimeValue:1 presentValue date=(125,6,1,1),time=(1,1,1,1)
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:__call__(PropertyReference) () {}
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:    - vanilla
DEBUG:bacpypes3.constructeddata.Sequence:(bacpypes3.basetypes.PropertyReference).__init__ None {}
DEBUG:bacpypes3.constructeddata.Sequence:    - elements: {'propertyIdentifier', 'propertyArrayIndex'}
DEBUG:bacpypes3.constructeddata.Sequence:    - init_map: {}
DEBUG:bacpypes3.constructeddata.Sequence:    - propertyIdentifier set to None
DEBUG:bacpypes3.constructeddata.Sequence:    - propertyArrayIndex set to None
DEBUG:bacpypes3.constructeddata.Sequence:__setattr__ 'propertyIdentifier' <PropertyIdentifier: present-value>
DEBUG:bacpypes3.constructeddata.Sequence:    - propertyIdentifier casting call: <class 'bacpypes3.primitivedata.PropertyIdentifier'> != <class 'bacpypes3.primitivedata.PropertyIdentifier'>
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:__call__(DateTime) ('date=(125,6,1,1),time=(1,1,1,1)',) {}
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:    - vanilla
DEBUG:bacpypes3.constructeddata.Sequence:(bacpypes3.basetypes.DateTime).__init__ None {}
DEBUG:bacpypes3.constructeddata.Sequence:    - elements: {'time', 'date'}
DEBUG:bacpypes3.constructeddata.Sequence:    - init_map: {}
DEBUG:bacpypes3.constructeddata.Sequence:    - time set to None
DEBUG:bacpypes3.constructeddata.Sequence:    - date set to None
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:__call__(WritePropertyRequest) () {'objectIdentifier': (<ObjectType: datetime-value>, 1), 'propertyIdentifier': <PropertyIdentifier: present-value>, 'propertyValue': <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>, 'destination': <IPv4Address 192.168.1.124>}
DEBUG:bacpypes3.constructeddata.SequenceMetaclass:    - vanilla
DEBUG:bacpypes3.constructeddata.Sequence:(bacpypes3.apdu.WritePropertyRequest).__init__ None {'objectIdentifier': (<ObjectType: datetime-value>, 1), 'propertyIdentifier': <PropertyIdentifier: present-value>, 'propertyValue': <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>}
DEBUG:bacpypes3.constructeddata.Sequence:    - elements: {'objectIdentifier', 'propertyValue', 'propertyArrayIndex', 'propertyIdentifier', 'priority'}
DEBUG:bacpypes3.constructeddata.Sequence:    - init_map: {'objectIdentifier': (<ObjectType: datetime-value>, 1), 'propertyValue': <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>, 'propertyIdentifier': <PropertyIdentifier: present-value>}
DEBUG:bacpypes3.constructeddata.Sequence:    - objectIdentifier casting call: (<ObjectType: datetime-value>, 1), element: <class 'bacpypes3.primitivedata.ObjectIdentifier'>
DEBUG:bacpypes3.constructeddata.Sequence:    - propertyValue casting call: <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>, element: <class 'bacpypes3.primitivedata.Any'>
    <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>
DEBUG:bacpypes3.constructeddata.Any:cast <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>
    <bacpypes3.basetypes.DateTime object at 0x7fce669a4190>
DEBUG:bacpypes3.constructeddata.Sequence:(DateTime).encode
DEBUG:bacpypes3.constructeddata.Sequence:    - date, <class 'bacpypes3.primitivedata.Date'>: None
write error: date is a required element of DateTime | Error type : <class 'AttributeError'>

JoelBender, 2025-03-08:
For your request to be successful there are a few changes to make.  First, because the value that you want to write will have embedded spaces, you need to change this:
```
> write 192.168.1.124 datetimeValue:1 presentValue date=(125,6,1,1),time=(1,1,1,1)
```
To this:
```
> write 192.168.1.124 datetimeValue:1 presentValue "2025-6-1 mon 01:01:01.01"
```
The rest is an enhancement.

The primitive data types like `ObjectIdentifier` are written so that if `x = ObjectIdentifier('device,1')` then `x == ObjectIdentifier(str(x))`, which is to say that you can round-trip a value to a string and back and get the same value, just like `12 == int(str(12))`.

For that to work with the `DateTime` class, in [this code](https://github.com/JoelBender/BACpypes3/blob/9322108b59acf39342a28f05ff685237b698f9e2/bacpypes3/basetypes.py#L1938-L1950) the `cast()` method that should detect it is being passed a string that would be the result of the `__str__()` method.  If you build a `DateTIme` object in code like you tried, the output would be the string value above.

As long as I'm in the business of parsing, I think that if you would want the time to be optional and default to midnight so `DateTime(date="2025-06-01")` would act like this:
```
>>> DateTime(date="2025-06-01", time="0:0:0.0").debug_contents()
    date = (125, 6, 1, 7)
    time = (0, 0, 0, 0)
```

But wait!  There's more!  There is a bug in `Time`:
```
>>> from bacpypes3.primitivedata import Time
>>> Time("01:01:01")
(1, 1, 1, 0)
>>> Time("01:01:01.1")
(1, 1, 1, 10)

>>> Time("01:01:01.01")
(1, 1, 1, 10)
>>> Time("01:01:01.001")
(1, 1, 1, 10)
>>> Time("01:01:01.0001")
(1, 1, 1, 10)
```

JoelBender, 2025-03-08:
Check out the `71-...` branch and let me know how it works for you.  Note that the strings that it can parse are not as flexible as the `Date`.

cadam01, 2025-03-10:
The branch 71 worked!  On the remote side, I a received a nice DateTime object and was able to test my write property handler.  I agree with the parsing when time is absent from the DateTime string.  I will be reviewing the 71 branch changes so I learn.

Thanks for your help Joel.

closed 2025-03-11 by JoelBender