JoelBender/BACpypes3 #73: Local transport property not set when trying the read-batch.py example

m2b, 2025-03-12:
I initialize the app from arguments same way as I do with the discovery example.  I have that one working quite well.  This one fails to retrieve any values (no response) for each item in the list (key,device_address,object_instance,property).

The address I pass for the devices are in the following form 
1000:0xc0a808730001.  I got that from the .pduSource of the IamRequest objects returned during discovery.
I also tried 1000:1 since I now that the device is 1 in network 1000.  I figure that the long hex string (ip=c0808730,network device address=0001) is a combination of the IP address of the MSTP-TCP/IP router that the device is connected to.

I can read the same properties, one by one, using app.read_property(device_address, object_identifier, prop). 

What am I doing wrong?

  File "bacpypes3/comm.py", line 135, in request
    await self.elementService.sap_indication(*args)
  File "bacpypes3/netservice.py", line 1164, in sap_indication
    await adapter.process_npdu(npdu)
  File "bacpypes3/netservice.py", line 498, in process_npdu
    await self.request(pdu)
  File "bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "bacpypes3/ipv4/service.py", line 300, in indication
    await self.request(lpdu)
  File "bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "bacpypes3/ipv4/bvll.py", line 228, in indication
    await self.request(pdu)
  File "bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "bacpypes3/ipv4/service.py", line 63, in indication
    await self.multiplexer.indication(self, pdu)
  File "bacpypes3/ipv4/service.py", line 94, in indication
    await self.request(pdu)
  File "bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "bacpypes3/ipv4/__init__.py", line 251, in indication
    self.local_transport.sendto(pdu.pduData, pdu_destination)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendto'

JoelBender, 2025-03-13:
> I figure that the long hex string (ip=c0808730,network device address=0001) is a combination of the IP address of the MSTP-TCP/IP router that the device is connected to.

No, check out the [address forms](https://bacpypes3.readthedocs.io/en/latest/gettingstarted/addresses.html) page in the documentation.  That string `1000:0xNNNN...` means it is the address of a device on the "other side" of a BACnet router to network 1000, at something more like a VLAN or Ethernet address "0c:80:87:30:00:01" which according to Google: The MAC address prefix "0c:80:87" is associated with Broadcom Corporation, a company known for its semiconductor and networking products.

The fact that you don't have a `local_transport` socket connection means that the address given to the application via a settings file, environment variable, or `--address` parameter isn't correct.  Somewhere in that traceback is probably something like "OSError: [Errno 49] Can't assign requested address".

If this was an exception with the `broadcast_transport` It could also be that the unicast address is fine (this machine happens to be 10.0.1.24) but it can't figure out the size of the network it is running on (this one happens to be 10.0.1.0/24 in CIDR notation) so attempts to send out broadcast messages like Who-Is-Router-To-Network are failing.

m2b, 2025-03-15:
Thanks for the response and clarification on the addressing scheme.  The issue is not that the CIDR is incorrect for the local address.  As, I mentioned, it all worked for discovery with the same local address.    I recommend you look at this slightly modified read-batch.py which reads from a file instead of stdin.    Everything worked fine if I called who_is from the app which initialized the device-cache and set the local_transport.  Comment and ucomment in the attached file to be able to see what I mean.

Otherwise, for the batchread.py objects, the indication method of the IPv4DatagramServer object is being called before self.local_transport is set in the set_local_transport_protocol method.  I don't know why but it happens.  The only hack I can come up with is using and event and waiting for the local_transport event to be set.  Here is what suggest inside of bacpypes3.ipv4.__init__

[image]

[image]

In the process of debugging I found that the batchread.py does not handle well the case of device as object-type.   It passes a tuple while ReadWritePropertyServices expects a sting or a ObjectIdentifier.    Here is what I suggest inside of batchread.py.

[image]

Attached files (.txt vs .py)

[read-batch.txt](https://github.com/user-attachments/files/19265744/read-batch.txt)

[batchread.txt](https://github.com/user-attachments/files/19265749/batchread.txt)

[__init__ .txt](https://github.com/user-attachments/files/19265775/__init__.txt)

JoelBender, 2025-03-17:
The `_local_transport_ready` fix has been added to [this branch](https://github.com/JoelBender/BACpypes3/tree/73-local-transport-property-not-set-when-trying-the-read-batchpy-example), and added your CSV sample application.  The "handle well the case of device as object-type" bug has been recently [fixed](https://github.com/JoelBender/BACpypes3/commit/a609e1708bdc19f328ca463e63c22f4de839ff8f).

Thank you for the extra work to find this bug!  It is bizarre that the task would complete but not be "ready".  Check out this branch and if it works I'll merge it in.

cyril-accenta, 2025-03-18:
Hi !
I've used the patched version of the file `bacpypes3/ipv4/__init__.py` and in about 25% of cases I encounter this error by just using a read_property_multiple : 
```
ERROR    asyncio:base_events.py:1821 Exception in callback _SelectorDatagramTransport._read_ready()
handle: <Handle _SelectorDatagramTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/lib/python3.12/asyncio/selector_events.py", line 1249, in _read_ready
    self._protocol.datagram_received(data, addr)
  File "/home/cyril/workspace/gitlab/generic_program/venv/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 54, in datagram_received
    asyncio.ensure_future(self.server.confirmation(pdu))
                          ^^^^^^^^^^^
AttributeError: 'IPv4DatagramProtocol' object has no attribute 'server'
```
Maybe there is a problem with the patch ?

JoelBender, 2025-04-01:
I received an email notification that **cyril-accenta** left a comment, but for some reason it vanished:

> I've used the patched version of the file bacpypes3/ipv4/__init__.py and in about 25% of cases I encounter this error by just using a read_property_multiple :

```
  File "/home/cyril/workspace/gitlab/generic_program/venv/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 54, in datagram_received
    asyncio.ensure_future(self.server.confirmation(pdu))
                          ^^^^^^^^^^^
AttributeError: 'IPv4DatagramProtocol' object has no attribute 'server'
```

> Maybe there is a problem with the patch ?

In this case, there could be another problem.  The `set_local_transport_protocol()` method is supposed to be called when the `retrying_create_datagram_endpoint()` has completed.  If there is a packet that is received after the the protocol object is created but before the `self.local_protocol.server = self` statement is executed (line 187), then there will be no server object.

The patch could be this:
```
        # drop incoming packets received before set_local_transport() called
        if (server := getattr(self, "server", None)):
            asyncio.ensure_future(server.confirmation(pdu))
```

Comments welcomed.

m2b, 2025-04-02:
Looks good to me.

ohshazbot, 2025-08-12:
Just to comment, I had an issue with the last release due to this stack:
```
future: <Task finished name='Task-9' coro=<ApplicationServiceElement.request() done, defined at /home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py:132> exception=AttributeError("'NoneType' object has no attribute 'sendto'")>
Traceback (most recent call last):
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py", line 135, in request
    await self.elementService.sap_indication(*args)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/netservice.py", line 1164, in sap_indication
    await adapter.process_npdu(npdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/netservice.py", line 498, in process_npdu
    await self.request(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/ipv4/service.py", line 300, in indication
    await self.request(lpdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/ipv4/bvll.py", line 228, in indication
    await self.request(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/ipv4/service.py", line 63, in indication
    await self.multiplexer.indication(self, pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/ipv4/service.py", line 94, in indication
    await self.request(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/comm.py", line 59, in request
    await self.clientPeer.indication(pdu)
  File "/home/onboard/.local/share/virtualenvs/onboard-bacnet-H5Uo4XkS/lib/python3.13/site-packages/bacpypes3/ipv4/__init__.py", line 251, in indication
    self.local_transport.sendto(pdu.pduData, pdu_destination)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendto'
```

I believed this to be attributed to https://github.com/JoelBender/BACpypes3/blob/0b5c12c3a8f9cdf0226a63dc3e711b11dfd696d2/bacpypes3/ipv4/__init__.py#L227-L234 because I was observing `set_local_transport_protocol` not present in the tasks since it was a chained response that did not become present until after the create_datagram_endpoint completed (https://github.com/JoelBender/BACpypes3/blob/0b5c12c3a8f9cdf0226a63dc3e711b11dfd696d2/bacpypes3/ipv4/__init__.py#L117-L119). I may not be correct in my analysis, but the [branch you posted](https://github.com/JoelBender/BACpypes3/tree/73-local-transport-property-not-set-when-trying-the-read-batchpy-example) does appear to address the issue.

Is there anything else you would need done to get that released to main?

ChristianTremblay, 2025-08-25:
> I received an email notification that **cyril-accenta** left a comment, but for some reason it vanished:
>
>
> ```
>   File [...]

@JoelBender 
I have seen this error in BAC0 script while creating a second BAC0 instance inside a `async with` creating the main BAC0 instance.  As soon as I encapsulated the second instance inside its own `async with`, error vanished. I suspect it is related to some kind of timing in the creation of instances, and it is possible that the script was giving the task to asyncio and went too fast to the other lines of the script.

My `async with` mechanism probably act as a guard, preventing the script to go too fast and wait for the creation of the instance, before continuing. Which is a good thing.

### Working script
```python
import asyncio
import random

from bacpypes3.primitivedata import Real

import BAC0
from BAC0.core.devices.local.factory import analog_value
from BAC0.scripts.script_runner import run

bacnet = None

async def main():
    async with BAC0.start() as bacnet:
        async with BAC0.start(port=47809, deviceId=123) as device:

            new_obj = analog_value(presentValue=0)
            new_obj.add_objects_to_application(device)

            # From Server
            dev_av = device.this_application.app.get_object_name("AV")
            print(dev_av.covIncrement)

            # From client
            ip = device.localIPAddr.addrTuple[0]
            boid = device.Boid
            bacnet._log.info("Defining device with poll 0 so the AV won't get updated")
            dev = await BAC0.device(f"{ip}:47809", boid, bacnet, poll=0)
            av = dev["AV"]
            bacnet._log.info("Subscribing to AV")

            def my_point_callback(property_identifier, property_value):
                print(f"CALLBACK {property_identifier}: {property_value}")

            await dev["AV"].subscribe_cov(lifetime=90, callback=my_point_callback)

            while True:
                dev_av.presentValue = Real(random.randint(1, 100))
                bacnet._log.info(f"Setting AV to {dev_av.presentValue}")
                await asyncio.sleep(5)

if __name__ == "__main__":
    run(main, bacnet)
```

### Script creating the error
```python
import asyncio
import random

from bacpypes3.primitivedata import Real

import BAC0
from BAC0.core.devices.local.factory import analog_value
from BAC0.scripts.script_runner import run

bacnet = None

async def main():
    async with BAC0.start() as bacnet:
        device = BAC0.start(port=47809, deviceId=123)

        new_obj = analog_value(presentValue=0)
        new_obj.add_objects_to_application(device)

        # From Server
        dev_av = device.this_application.app.get_object_name("AV")
        print(dev_av.covIncrement)

        # From client
        ip = device.localIPAddr.addrTuple[0]
        boid = device.Boid
        bacnet._log.info("Defining device with poll 0 so the AV won't get updated")
        dev = await BAC0.device(f"{ip}:47809", boid, bacnet, poll=0)
        av = dev["AV"]
        bacnet._log.info("Subscribing to AV")
        await dev["AV"].subscribe_cov(lifetime=90)

        while True:
            dev_av.presentValue = Real(random.randint(1, 100))
            bacnet._log.info(f"Setting AV to {dev_av.presentValue}")
            await asyncio.sleep(5)

if __name__ == "__main__":
    run(main, bacnet)

```