JoelBender/BACpypes3 #79: Can bacpypes3 create local objects of schedule and enent?

fanchunyong, 2025-03-24:
[image]
I see there are schedule.py and event.py in bacpype3, so does bacpype3 support local objects for schedule and event?

bbartling, 2025-03-24:
You want BACnet server points, right?...for schedule and event

fanchunyong, 2025-03-24:
@bbartling Yes, I want to create server points, what should I do?

cyril-accenta, 2025-03-24:
I'm not sure I got the question but just in case we are using this code with great success : 

```python
    ipv4_address = IPv4Address('127.0.0.1', port=port)
    device = DeviceObject(
        objectIdentifier=("device", 1),
        objectName="server",
        vendorIdentifier=999
    )
    new_server = NormalApplication(device, ipv4_address)
    new_server.add_object(ScheduleObject(
        objectIdentifier=("schedule", 1),
        objectName="Test-1-Schedule",
        presentValue=Integer(1),
        weeklySchedule=[
                           DailySchedule(
                               daySchedule=[
                                   TimeValue(time=(8, 0, 0, 0), value=Integer(1)),
                                   TimeValue(time=(14, 0, 0, 0), value=Null(())),
                                   TimeValue(time=(17, 0, 0, 0), value=Integer(0)),
                               ]
                           ),
                       ] * 7,
        scheduleDefault=Integer(0),
    ))
```

This code is placed in an async function, run in the loop with a run_until_complete then the loop is launched with a run_forever and voilà, your server will serve a schedule object.

bbartling, 2025-03-24:
Maybe try experimenting on `mini_device.py`

https://github.com/JoelBender/BACpypes3/blob/main/samples/mini_device_revisited.py

Just try running it and then modify as needed. I am curious about this as well it would be neat to have an enhanced mini_device py that can do a schedule objects.

The mini device example code should get another make over as well for points you can do a default units type and bacnet point description....something like this.

```
        self.temp_av = AnalogValueObject(
            objectIdentifier=("analogValue", 1),
            objectName="oa-dry-bulb",
            presentValue=0.0,
            statusFlags=[0, 0, 0, 0],
            covIncrement=1.0,
            units="degreesFahrenheit",
            description="Web Weather Outside Air Dry Bulb Temp",
        )
```

I have a hobby project that is based off of the mini device py

https://github.com/bbartling/bacnet-web-weather-linux/tree/develop

If you had a working code base for another enhanced mini device py to demo a schedule object please post the code! 

Hope this helps.

bbartling, 2025-03-24:
https://github.com/bbartling/bacnet-web-weather-linux/blob/278a0ec855f3240955adc06292915e2407dca205/app/main.py#L63C1-L63C25

That is the web weather hobby app for bacnet server

bbartling, 2025-03-29:
Am trying to PR this into the repo. Curious on what else would you be looking for? https://github.com/JoelBender/BACpypes3/blob/cc04e68c337a1cc9f95c88f45c68b50e52a98f43/samples/mini-device-revisited.py