JoelBender/BACpypes3 #83: There was a problem implementing GetAlarmSummaryRequest

jhpark010, 2025-04-16:
I'm currently using bacpypes3 version 0.0.102, and I have successfully implemented the event information retrieval feature using the GetEventInformationRequest class.

However, while developing the alarm summary retrieval feature using the GetAlarmSummaryRequest class, I encountered an error.
On the client side, I wrote the following code

```
async def send_get_alarm_summary(self):
    get_alarm = GetAlarmSummaryRequest()
    get_alarm.pduDestination = Address("IP/24")

    try:
        response = await self.request(get_alarm)

~
```

However, the following error occurred, and after tracing the source of the issue,
I found that it seems to be caused by the GetAlarmSummaryRequest class definition in the library's apdu.py file,
specifically due to the _order = () declaration.

**ERROR**
```
C:\Lib\site-packages\bacpypes3\constructeddata.py", line 442, in decode
    raise RuntimeError("sequences must be ordered")
RuntimeError: sequences must be ordered
```

**adpu.py Line 1340**
```
@register_confirmed_request_type
class GetAlarmSummaryRequest(ConfirmedRequestSequence):
    service_choice = ConfirmedServiceChoice.getAlarmSummary
    _order = ()
```

Is it possible that I should be using a different method instead of GetAlarmSummaryRequest() to retrieve alarm information?
If so, I would appreciate it if you could suggest a solution or provide guidance on how to resolve this issue.

JoelBender, 2025-04-26:
This is a "guard" in `Sequence` to make sure that `_order` has been provided in the subclass definition.  The default of an empty tuple is correct when there are no elements.

@jhpark Check out this branch and if you agree I'll merge it.

jhpark010, 2025-04-29:
Hello,
When I executed the code using the modified branch, the error no longer appeared during the normal encoding process.
However, the same error occurred during decoding. To work around this, I manually commented out lines 441–442 in constructeddata.py, and after that, I was able to execute the GetAlarmSummaryRequest successfully.

I would like to ask if commenting out lines 441–442 in constructeddata.py was an appropriate or advisable approach.