JoelBender/BACpypes3 #84: Migrating from bacypes.analysis to bacpypes3.???

acedrew, 2025-04-20:
Hey @JoelBender trying to figure out how to recreate the functionality in bacpypes.analysis.decode_packet in bacpypes3. I see samples/apdu-hex-decode.py but I haven't figured out how to easily walk the tag list back into a full featured request or response object like decode_packet did, if the answer is just walk through it layer by layer loading each appropriate type, I'll take a stab at it, but I wanted to make sure I wasn't missing some more efficient approach that bacpypes3 structure might allow.

JoelBender, 2025-04-25:
There's no real magic, but I've got a starting point with [python-libpcap](https://python-libpcap.readthedocs.io/en/latest/) that has an easy starting point.

JoelBender, 2025-04-25:
Check out this branch and let me know if it works for you.  I'm considering (1) moving it to a **bacpypes3-analysis** project and/or (2) having it decode into JSON blobs so `decode_packet()` has the same signature as `decode_ethernet()`, or some other combinations of content that is closer to the output of `tshark -Tjson`.

It'd be fun to roll that into a Streamlit/Gradio application for browsing around with the other who-is/i-am alignment stuff.

acedrew, 2025-04-25:
I think all of my problems doing this myself were because I was trying to use a different library for the pcap management. IMHO, this basic bytes->bp3 object makes sense to be in this codebase. Anything else maybe a separate project. I'm working on integrating semantic network metrics into grasshopper. This is perfect for now.

acedrew, 2025-04-26:
Also, for what it's worth, I don't really want JSON output, I want the native request and response objects. Especially if we get type stubs :) 
I'm still struggling to get everything I need, I see settings.route_aware, but when set to true, I'm having trouble figuring out where to find the underlying routed device in the returned APDU, it looks like it should be at apdu.pduSource.addrRoute but I'm getting None with my test packet. If I switch route_aware back to False, I get the MSTP address, but no information about where it came from. I'm thinking maybe decode_packet is throwing away context after it casts everything that I really want for my analysis. How would you feel about a more functional implementation that would collect all the basetypes as it peeled back the layers? Much like the underlying packet representations in libpcap or scapy for instance. Basically I want the NPDU and APDU info at the end, instead of the just the APDU with overwritten sources and destinations based on the route_aware flag.

JoelBender, 2025-04-26:
Ok, try this version.  All of the individual pieces are collected together into a "frame".  There are still some content that is carried forward from one layer to the next, like `pduExpectingReply`, and some `data` content is "consumed" by the next layer up.  Understanding the PDU source and destination is now up to the user, just like BACnet :-).

acedrew, 2025-04-26:
I'll take a look, I was working on something similar, I'm not asking you to do all the work! Just want to make sure I'm not doing something that isn't worth doing.

JoelBender, 2025-04-26:
I was using JSON to stream into a MongoDB capped collection for a dashboard, I can use some combination of `asdict()` and `sequence_to_json()` and cover that use case.  It's absolutely worth doing!

acedrew, 2025-04-26:
Oh cool, yeah we're aggregating it into metrics for dashboards so we can keep them around for longer and then keeping the pcaps as the raw source for detailed analysis in a rolling s3 bucket archive. I bet opensearch would be another great place to stick this stuff. The RDF model will just serialize the aggregates.